kwindrem/GuiMods #192: DIY / Code Guidelines?

spacecabbie, 2023-09-16:
Was wondering if there is anything like this I like to try and make some changes for my self.

kwindrem, 2023-09-16:
There is a document in SetupHelper that explains how to make a "package" compatible with the SetupHelper install mechanism.

I do not have anything regarding where to make specific changes to the system. 

All (or most) GUI modifications are made in the QML code: /opt/victronenergy/gui/qml. 

Other parts of the system typically use a "service" that runs as a separate background program. These are located in /opt/victronenergy in separate directories for each part of the system. Most services are written in Python.

Some of my mods (ExtTransferSwitch and ShutdownManager, and also PackageManager which is part of SetupHelper) add their own services.

dbus is used to communicate parameters between parts of the system. This is all explained here:

https://github.com/victronenergy/venus/wiki/dbus

Hope this helps.

closed 2023-09-16 by kwindrem

spacecabbie, 2023-09-16:
Thank you this is good information although didn't realize you actually run separate services that makes it more complex then i though i just hoped to grab an extra value here and there. Still wil have a look thanks so much for the quick guide

kwindrem, 2023-09-16:
GuiMods currently does not run a separate service.