kwindrem/GuiMods #196: request please

sarah1331sarah, 2023-10-21:
thank you for all your efforts its really appreaciated 

is there any chance to have a letter at the side of the multiplus "led's" i know the top ones are "on" and inverter on
the left ones i remember but the right i always forget (hopefully typing this will be the revision i need)

thanks for eveything again 

regards paul

kwindrem, 2023-10-23:
I've never actually looked at those lights mainly because I do not know what they mean. Having some indication of the light's purpose would make them much more useful. I'll look into it. Space is tight so I'll see what's possible.

kwindrem, 2023-10-28:
I am playing with ideas for LED labels but it makes for a very busy layout:

[image]

I wonder if only the active LEDs should display their labels.

I"m not all that happy with the single letters for charger mode but the same information is generally shown in the tile's title ("Float" in the case shown above").

Single letter labels for inverting, overload, low battery and over temperature might be too cryptic.

Thoughts?

sarah1331sarah, 2023-10-28:
I see what you mean it looks good I mean brill but there is  something not 100% about it.

when active may be a good idea but would it ;
Only come on while the led is lit I.e when flashing would the “Tmp” flash too
Would people be thinking what’s the others for while in good working order

kwindrem, 2023-10-28:
Didn't realize the LEDs flash. I know they do on the actual device but do they on the GUI also?

Thinking about replacing the LEDs with text that changes color when active. Gray/black when inactive, green, yellow or red when active based on the LED colors. That would provide room for more meaningful text: AC In, Bulk, Abs, Float, Inverting, Overload, Low Bat, Over Tmp. Might need to abbreviate these some and there won't be a language 
translation.

sarah1331sarah, 2023-10-28:
This is a good idea. yes it flashes on the gui also. You will have to proto type it it may take away from the iconic look removing the LED’s

kwindrem, 2023-10-28:
[image]

Here's another cut. The text fields will use the same on colors and blink capability as the LEDs. I've set the off color to black to make them more readable

Comments?

sarah1331sarah, 2023-10-28:
I like it buddy would be good to hear any other feedback I feel like there’s only us that care

sarah1331sarah, 2023-10-29:
~ ac in 
:| ~ inverter 
Battery steady and swap with red line rev counter when overload
Battery and a swap /flashing empty battery when low  
Thermometer and red high thermometer when flashing

kwindrem, 2023-10-29:
I'm sorry, I'm not understanding your last comment.

sarah1331sarah, 2023-10-29:
forgive my amature skills (should get my 12yr old daughter to help me)
[image]

kwindrem, 2023-10-29:
OK, got it: icons. Good idea. Eliminates the issue with language translation.

I'll need to see what kind of room I can make. With the LEDs stacked vertically, the thermometer icon would need to be sideways and might loose it's recognizability.

virtuvas, 2023-10-31:
> [image]
> 
> Here's another cut. The text fields will use the same on colors and blink capability as the LEDs. I've set the off color to black to make them more readable
> 
> Comments?

fwiw, I very much like that K.!
keeps the original logic, helps as I can never remember what each led means and I have to switch to the detailed Multi screen to see the full text. 
Not so sure you'll manage to get a decent use of the avail space using icons. It's probably going to be more confusing and way too "busy" (imho)

cheers

V.

jsalbre, 2023-10-31:
Agreed on the comments below. The text-as-LED seems like the best version.Jeremy

kwindrem, 2023-10-31:
Done - used text (not icons)

closed 2023-10-31 by kwindrem