kwindrem/GuiMods #199: White screen in Remote Console after latest (automatic) update

R0Li84, 2023-11-06:
Hello, I observed that there was an automated update for GUI mods tonight. However, after the update was installed, I encountered an issue where I can only see a white screen when accessing the remote console.

I took the initiative to research this problem and found that it had occurred in the past as well. Could you please investigate and resolve this issue? I am using Venus OS on my Raspberry Pi, and I have installed the latest official version. Perhaps this issue could be fixed with another automated update?

[image]

chelle-hine, 2023-11-06:
I have the same issue on v3.10 with guimods. I'm currently looking at previous occurrences, for methods to recover it, but if another auto update will likely fix it soon, I'll wait.

the-alpine, 2023-11-06:
I have exact same problem.  White screen on the display, but still updating via VRM.

Stefan66AB, 2023-11-06:
Same problem.
--> I installed a firmware update via VRM portal to Venus OS 3.12. Then it worked again...

kwindrem, 2023-11-06:
I'll fix the white screen (it's a black screen in dark mode BYW) which appears to only be for v3.10. For now, upgrade to v3.12.

the-alpine, 2023-11-06:
I update the firmware via VRM as suggested by Stefan and that did the trick.

chelle-hine, 2023-11-06:
I've now updated to v8.18, which cleared the white screen, enabled two way communication, and updated the OS to 3.12 via the VRM, and still OK.

Thanks for your quick help @kwindrem

felizcortez, 2023-11-06:
I am also having the same problem with the white screen. Running cerbo gx Venus large and auto updates from guimods.  How do you update the firmware via vrm?  I'm not sure whether I have two way communication enabled.

Will there be an auto update that fixes this?

chelle-hine, 2023-11-06:
VRM update instructions/video here....

You can see if you have two way comms enabled on the same screen of VRM

https://www.victronenergy.com/live/vrm_portal:remote-firmware-update

kwindrem, 2023-11-06:
GuiMods v8.18 should fix the white/black screen. Should work with Venus OS v3.10

felizcortez, 2023-11-06:
I was on 3.01 and updated to 3.12 via VRM and that seems to have fixed the issue.

felizcortez, 2023-11-06:
> I was not aware of an issue with v3.01 as well. Fixed in GuiMods v8.20 available now

Awesome. Thank you.  For the quick help resolving this issue.

jsalbre, 2023-11-08:
I’m on 3.20-18 with guimods 8.20 and I’m getting the blank screen now as well. I’m not sure how to recover from this since I can’t downgrade via VRM.

jsalbre, 2023-11-08:
I’m on 3.20-18 with guimods 8.20 and I’m getting the blank screen now as well. I’m not sure how to recover from this since I can’t downgrade via VRM.Jeremy

kwindrem, 2023-11-09:
If you can connect a serial terminal to the GX device, you may be able to install new firmware placed on a USB stick from the command line:

You may need to reimage. There are instructions for Cerbo here:

https://github.com/kwindrem/SetupHelper/issues/56

On a Pi, just reimage the SD card:

/opt/victronenergy/swupdate-scropts/check-updatess.sh -offline -force -update

jsalbre, 2023-11-09:
Unfortunately I’m traveling right now so I don’t really have any way to do the whole serial cable dance. I noticed if I connect to the remote console via VRM I still get the white screen, but the CCGX style buttons are still visible. On the chance the GUI is still functional, just not visible I was thinking about trying to blindly go through the menu and uninstall guimods, or reboot into the 2nd boot partition, but I’m not sure what key press pattern I’d need.Any chance anyone might be able to count those steps out for me?

kwindrem/SetupHelper#56

jsalbre, 2023-11-09:
To clarify, I don’t seem to have the problem some are where the Cerbo won’t boot. I can access nodered and its dashboard, and the Cerbo is connecting to VRM and sending data. I can even connect to ssh, though unfortunately I don’t have the password I created handy.

kwindrem, 2023-11-09:
If you have ssh access you can get to logs and even uninstall packages from the command line.

jsalbre, 2023-11-10:
I couldn’t get ssh, but I was able to overnight a serial interface and connect to the cerbo that way. I’ve got it booted into the backup FS, and it’s all working now. If I’m understanding the notes in issue #55 this shouldn’t reoccur with future updates, correct?Thanks!

kwindrem, 2023-11-10:
SetupHelper has been modified to insure there is sufficient space before proceeding with any package install. This affects GuiMods and any other package utilizing SetupHelper for the install. So yes, I believe the issue has been resolved.