kwindrem/GuiMods #200: Guimods 8.xx unusable on CCGX

GeoZBee, 2023-11-08:
I've been using Guimods for quite some time now. And while stability of the CCGX with guimods installed has been a constant issue, it was manageable, causing a few reboots a week or whenever navigating through the settings screens "too much".

Recently, I updated my CCGX (using the package manager) from 7.7 to 8.17 (along with setuphelper, to 5.15, first).
The installs seemed to go well enough, and I saw the stated updates.
But from that point forward, the CCGX has became very unstable.
Instead of rebooting whenever I used the settings screens "too much", it is now happening within a few clicks of navigating the settings screen. 
And instead of the usual handful of reboots a week, it is now rebooting on its own several times a day.
The UI is also very sluggish and not responsive.

I tried updating to 8.18 and saw no difference. 8.19 came out shortly after and I installed that with no difference. 8.20 came out, and I installed it and that added a _slight_ improvement to stability -- instead of it rebooting every few minutes, it is now rebooting several times a day, entirely on its own.

That is the way it sits now while I decide what to do next. 
I figured I would post this here, first, before trying to cobble together a way to back out to 7.7 or some other more stable build.

I do really like this mod, and I'm sure it runs like a champ on more robust hardware. But on a CCGX, at least for me, it has become completely unusable.

GeoZBee, 2023-11-09:
Updating the CCGX firmware to 3.12 (from ~2.84) has made this a _bit_ more stable. 
It's been about 36H since I updated, and while the CCGX still reboots on its own, it has now only done so twice.
It still, however, reboots whenever navigating around more than a handful of the settings screens, or when switching to the details panes of individual objects from the flow screen and back "too many times".

kwindrem, 2023-11-09:
I'll admit that the CCGX is very slow when navigating GuiMods objects, especially the overview pages, and that is with no devices connected to it. I have not ever had it crash and require a reboot.

Disabling overview pages that you don't need helps since these are all loaded when switching to the pages view (from menu).

I know this is an expensive option but upgrading to Cerbo or Eranko might be your only option, or stop using GuiMods.

GeoZBee, 2023-11-15:
For what it's worth, it doesn't _crash and require a reboot,_ it reboots entirely on its own, untouched and just sitting around with zero interaction. Sometimes that's within a few hours, or (in the past) a few days.
Navigating any of the screens within the settings significantly exacerbates this process, and the unit will reboot on its own, mid-navigation.

With 3.12, I've got it stable-ish enough for now. And with all the features offered by Guimods make it a tolerable tradeoff, even if not ideal.
It is highly unlikely, however, that I will risk applying any future Guimods updates; which is too bad, because now that I'm used to them, I like the updates in 8.x. 
But, I also _do_ get it. The CCGX isn't exactly a sexy new hotrod, and a Cerbo is probably the long-term fix.