kwindrem/GuiMods #201: AC in data missing line voltage and used amps

GeoZBee, 2023-11-08:
After updating to 8.20 (from 7.7), the AC/Grid/shore in data box (simple flow view) now only shows watts (at the top) and the rate limit in Amps (at the bottom) and nothing else.
It is missing the line voltage and used amps (as it showed before).
I have found nowhere to reactivate this data in settings, and I'm not sure if this is intentional, a bug, or I'm missing something. 

I can provided whatever additional data would be useful.

GeoZBee, 2023-11-08:
I decided to do some more experimenting.
I was running CCGX firmware 2.9X.
I updated to the latest firmware available (3.12).
That uninstalled GuiMods in the process.
After the update, I reinstalled guimods, again.
This issue has now since resolved, and the missing AC-in data is showing as expected.

kwindrem, 2023-11-08:
Do you know what specific version you were running before you upgraded to 3.12? I can check my code and even test with that version to see if there's an issue.

GeoZBee, 2023-11-09:
I'm pretty sure what was installed was actually 2.84 (not 2.9x).
I had downloaded a 2.9 image, but not installed it.

kwindrem, 2023-11-09:
I was not able to reproduce the issue on 2.84 which should be the same as 2.9x unless I missed a changed file in comparing versions. I don't have anything hooked up to my test system and the system in my travel trailer is difficult to revert to that old a version. This is not so say there might still be issues.

I'm going to close this for now since you have moved on.

closed 2023-11-09 by kwindrem