kwindrem/GuiMods #203: Battery detsil

kocki2, 2023-11-10:


kocki2, 2023-11-10:
current - consuming -> charging -- in state Charrging
current - supplying -> discharging -- in state Discharging
[image]
[image]

kwindrem, 2023-11-10:
I see your point. 

The consuming/supplying is consistent with other detail pages. This was adopted in stead of showing positive and negative signs to avoid confusion since a signed number is used inconsistently in the Victron ecosystem: power consumed by a load is a positive number; power supplied by the grid is a positive number.

Only for the battery do we label power flow as charge and discharge. and on this details page the same power transfer is labeled as both charging and consuming.

Let me give this some thought. If you have more input, I'm happy to listen.

kocki2, 2023-11-11:
Okay, I understand what you mean.
Maybe I have anoter point of view because I dön't have a home installation, but an installation on a sailing boat.
Because this is the Battery detail tab it would be more clear for me if the current value would be -1 A instead of 1A supplying and +2A instead of 2A consuming.
Actually I look at the current value A and then to state discharging / charging.
I would prefer +/- A without supplying / consuming.
But - of course - this is my point of view.
What do you and other users think about this?
Please tell me.
Regards, Gerd

kwindrem, 2023-11-11:
I don't have a problem with +/- as long as it is consistent. For example, AC loads should be a negative number since they are consuming power. But it isn't that way. However, Victron has a system where no sign loosely means normal. A source has an unsigned number when it is supplying power. A load has an unsigned number when it is consuming power. The DC System tile is unsigned power when loads exceed sources and negative if sources exceed loads.

There are many like you (and me) that understand positive and negative power and find +/- an intuitive mechanism for determining power flow. But it needs to be consistent: sources are positive and loads are negative. So AC Loads would present a negative number. That's technically correct but may not be understandable to users that don't fully understand the power flow in the system.

I have discussed this with several others and the supplying/consuming terminology was our best compromise.

kocki2, 2023-11-11:
Okay, I see your point, it's okay for me, I closed the issue.
Have a nice day,
Gerd

closed 2023-11-11 by kocki2