kwindrem/GuiMods #204: No fileset

thomasinaz41213, 2023-11-11:
I am still on Venus 3.10~37 and guimods is stuck on 8.10. it shows 8.21 available but says no fileset available for 3.10~37

kwindrem, 2023-11-11:
Venus OS v3.10 is released (and there is actually a v3.12 available). You should upgrade to that rather than staying on a release candidate (beta) version.

I routinely remove file sets for beta versions of Venus OS once the released version is available.

closed 2023-11-11 by kwindrem

thomasinaz41213, 2023-11-11:
I've been meaning to update, just haven't had the time. I've been extremely busy and I have a few other mods which have to be migrated manually so it takes a little while to do an update. Guess I better just bite the bullet and get to it.

kwindrem, 2023-11-11:
Sorry for the inconvenience. Yes, it is probably time. Or you can just stay on GuiMods 8.10. Only critical updates (according to the changes document) are for the generator interface and cool-down.