kwindrem/GuiMods #205: Looking forward to the next update ...

StupidMonkeyKev, 2023-11-18:
[image]

Looking forward to the improvements you can make to the new Victron screens - already missing the extra info, may even roll back the firmware to get my GuiMods back. 

Thanks for all your efforts!

kwindrem, 2023-11-18:
I have no plans to modify the new GUI.

GuiMods will work on the original GUI as long as it is available on the GX device.

Tiles in the new GUI will have "drill-down" pages (like GuiMods details pages) to show detail. The tiles will also contain gauges to indicate relative power.

StupidMonkeyKev, 2023-11-18:
No worries, I'll roll back the auto-update on my Cerbo to get my GuiMods back

kwindrem, 2023-11-18:
You can switch back to the standard  GUI in the Device & Language menu (very bottom). This is on v3.20~26 and I expect it will be that way until the new GUI is out of beta.