kwindrem/GuiMods #208: Cerbo GX Setup Helper

Sportie6423, 2023-11-22:
Hi Kwuindrem, Im having a problem with the "blindinstall". I follwed the steps, Format USB (Fat), download the file which i send to the USB, it shows up in the format of venus-data tgz, I eject the USB and then place the it in the Cerbo GX, then reboot and reboot again, But the product manager never shows up under the I/O. Ive tried even powering off the Cerbo and power cycling but to no avail. Ive read everything i can find on trouble shooting including your responses to others but cant get this to work. I did see where you could remote install it, but that rasberry PI etc seems a little much for me, I got lost reading that lol. Anyway if you have any ideas for me please help. 

Cerbo GX  Firmware 3.12

kwindrem, 2023-11-22:
I just tried a blind install on my test system and it worked. Only one reboot is necessary with the current firmware. My test system is a Raspberry PI, not Cerbo but there should be no difference.

To be sure, the venus-data.tgz file should be placed on the USB stick or SD card as-is. Do not unzip it.

Have you set up your system for root access?

Instructions here: 

https://www.victronenergy.com/live/ccgx:root_access

logs are in /data/logs. Of interest would be /data/logs/SetupHelper

Sportie6423, 2023-11-22:
Yes I think I did. I followed someone one YouTube to turn on the ssh (pic 1). Changed to super user and created a password. Then turn on ssh. I’m hoping that’s how to create root access. Sorry I’m a newbie to this. And as far as the tgz file, I downloaded and sent straight to the usb. I did notice once I tried the install and ejected it, it had a second file on the usb? Sent the pic 2. It only had the Venus-data.tgz file before the install was attempted. Any idea?

I don’t know how or where to find any logs without help. Sorry I’m sure you’re busy. I just can’t figure it out myself 
[image]
[image]

kwindrem, 2023-11-22:
The second file is a VRM backup database which is written to the first removable media device the GX device finds. If the connection to VRM is lost (no network), this file saves the records until VRM connection is established.

Log files are in /data/log. There are generally three log files of interest and you can access them with the following commands:

tail -50 /data/log/SetupHelper | tai64nlocal
tail -50 /data/log/PackageManager/current | tai64nlocal
tail -50 /data/log/gui/current | tai64nlocal

or you can copy the entire lot file to your host with something like:

scp root@venus.local:/data/log/SetupHelper .

The SetupHelper log file should contain messages from the blind install which is what I call the procedure involving venus-data.tgz.

venus-data.tgz should be a little under 100 KB. If it is much smaller, it probably didn't download from GitHub or to the USB stick properly.

That will place a copy of the SetupHelper log file in the current directory on your host.

You need some kind of terminal program (terminal on MacOS, or install putty on Windows. I don't use putty so there might be another way to transfer files than using scp.

Good luck.