kwindrem/GuiMods #209: Integrate Cerbo GX with data to/from a nearby Zigbee network?

kjel-siv, 2023-11-22:
I have recently joined this amazing community and are still discovering the features of my new Victron Cerbo GX with GuiMod and the Node Red software. 

In my boat I want to integrate the Victron Cerbo GX with my Zigbee network to be able to use e.g. the geofence feature in the Cerbo to set the alarm features in the Zigbee world, and vis-a-versa be able to show eg. the Temperature data from the Zigbee sensors on the Cerbo GX display.

Has anybody done this already or do anybody have a clue where to start this journey?

kwindrem, 2023-11-22:
Welcome to the Victron world.

What you are suggesting is not something GuiMods can accomplish. Zigbee integration will require some significant changes to Venus OS as it will affect many different sub-systems. Please post this as an question/idea in the Victron Community forum modifications section:

https://community.victronenergy.com/spaces/31/index.html

closed 2023-11-22 by kwindrem

kjel-siv, 2023-11-22:
Thanks for the swift reply and advice :-) 

Done!