kwindrem/GuiMods #210: Missing dbus-adc.conf file

RobDuthie, 2023-11-28:
Missing dbus-adc.conf file in latest update. from file location etc/venus/
As this installs the analog ports required for temap and tanks.

device iio:device0
vref 1.8
scale 4095

tank 0
tank 1
tank 2
tank 3
temp 4
temp 5
temp 6
temp 7

kwindrem, 2023-11-28:
GuiMods does not touch dbus-adc.conf so I'm not sure what you are saying.

RobDuthie, 2023-11-28:
Hi Kevin

Re missing dbus-adc.conf file,
When doing an update it needs to be updated sorry as when you access the 
analog port they are not there.
I manual add that file as to make analog ports show up for the stetson 
hat on the Raspi PCB.
As i had to help a few people over the matter on how to make analog 
ports work, just by adding the file changes it works again that's all.
Just need to add into the script the changes and add abus-adc.conf as 
attached

Regards

Rob Duthie
Manager
Radio Electronics Tech
Electrical Service Tech (A&B)
Electrician

*RD Electronics & Electrical & R&G Energy Systems ltd*
*Electronic Design & Service*
*Email: ***@***.***
*Email: ***@***.***
*Ph: *07 3456043
*Mobile: *0272898824
*Address: *11 Selwyn Rd, Rotorua,3010
New Zealand
device iio:device0
vref 1.8
scale 4095

tank 0
tank 1
tank 2
tank 3
temp 4
temp 5
temp 6
temp 7

kwindrem, 2023-11-28:
Rob, I'm still very confused. None of my packages touch the file you mention. Are you suggesting I SHOULD be creating/modifyng this file? Isn't that something the setup scripts for Stetson should be doing?

RobDuthie, 2023-11-28:
Hi Kevin

Don't worry i will modify it my self this is what i did when i updated 
the script file my self along time ago.
At the moment this file is blank in the venus folder etc/venus software 
no data in it so i do is just replace it.
and all the analog ports pop and work OK.
-- 

Regards

Rob Duthie
Manager
Radio Electronics Tech
Electrical Service Tech (A&B)
Electrician

*RD Electronics & Electrical & R&G Energy Systems ltd*
*Electronic Design & Service*
*Email: ***@***.***
*Email: ***@***.***
*Ph: *07 3456043
*Mobile: *0272898824
*Address: *11 Selwyn Rd, Rotorua,3010
New Zealand

RobDuthie, 2023-12-09:
Found out why added in a new file into the rcS.local file to copy across and all is ok now.
dbus-adc.conf file.