kwindrem/GuiMods #214: AC L2 output not shown on inverter details page

thomasinaz41213, 2023-12-12:
I have the MP2 2x120, currently connected to 50 amp split phase power. On the inverter details page there is >>> displayed where the L2 output should be. All other details screens show the correct value.
[image]

thomasinaz41213, 2023-12-12:
I just noticed on the inverter details page it says L2 included in L1 values, as if it was plugged into single phase power, which it is not. Here is a screen of the main display, you can see that it is clearly 50 amp split phase.
[image]

kwindrem, 2023-12-13:
What firmware version are you running?

How close together in time were these screen shots? I see 9.7 amps for L1 input current which should also be showing on the AC Input and AC Loads L2 in the overview page.

I'm seeing some coding errors but I'm thinking these would be showing L2 when it is not being used.

If you have terminal access, a screen shot of what dbus-spy is showing for the inverter, specifically:

/Ac/State/SplitPhaseL2L1OutSummed 

and

/Ac/State/SplitPhaseL2Passthru

thomasinaz41213, 2023-12-13:
Those screen shots were taken a few minutes apart, sorry about that. I took the second one after I started writing the post. Here are 2 shots within seconds of each other.
[image]

[image]

[image]

kwindrem, 2023-12-13:
Thanks. That will help sort this out.

The SplitPhaseL2 Passthru parameter shows L2 is passing through the unit unmodified so AC in and out for L2 should be the same -- and they are in the screen shot you provided.

As I think more about this, what the Inverter detail is showing is correct. The inverter does not have an L2 leg in the 120x2 units even when L2 power is passing through from AC in to AC out. I need to rethink how to show this so it is clear there is no L2 inverter by design.

The AC input an AC output detail pagess should display values properly. Can you verify that for me?

AC output should switch to L1 + L2 (aka L2 included in L1) when inverting (no shore power).

thomasinaz41213, 2023-12-13:
AC input and AC output details pages are showing correct info. I get what you are saying about the "Inverter Details" page not showing values for L2 since this is a single phase inverter, makes sense. How about showing something like "passthru" instead for when the inverter is connected to split phase? I think that would be more accurate?

thomasinaz41213, 2023-12-13:
Bottom of the inverter details page also says "L2 output values included in L1" which would be incorrect when connected to split phase and L2 is in passthru.

kwindrem, 2023-12-13:
I'll see if there is room for "pass thru" for L2 but it would be shown on every line so redundant and very busy. My current thought is to show only L1 in the table and have a note at the bottom that there is "no L2 inverter/charger" or something like that.

I ran into an additional problem that GuiMods does not handle L2 in the 120x2 Multi and Quattro models properly. I need another day to get something to test.

kwindrem, 2023-12-13:
I have a new implementation for the Inverter Detail.
First, there was a bug that showed <<< when L2 is in pass-thru not when the inverter is active.

I chose to leave the L2 info in place but changed L2 power to "none" when in pass-thru. I also added a message for the pass-thru mode:

[image]

[image]

Comments?

thomasinaz41213, 2023-12-13:
Looks good 👍

kwindrem, 2023-12-13:
v8.27 is released with these changes. Please look at the behavior of the AC input and output tiles in the flow page as well as their details pages and the Inverter details page. Inverting vs on shore power affect what is shown so it would be good to check all in both modes. Thanks.

Any issues, just reopen this issue.

closed 2023-12-13 by kwindrem

thomasinaz41213, 2023-12-13:
Looks good. Working as expected in both modes.

kwindrem, 2023-12-13:
Great. Thanks for testing it. Only so much I can do here with my single phase setup in my travel trailer.

thomasinaz41213, 2023-12-13:
Any time you need something tested or verified on a 2x120 system just give me a shout. I'd be happy to help.