kwindrem/GuiMods #215: GuiMods glitch - suddenly not working

lfhoward, 2023-12-12:
Dear Kevin,

At some point last night or this morning GuiMods had some kind of problem and is no longer functioning. Perhaps a bad auto-install? I used SetupHelper to remove it and reinstall, and rebooted after each change. I cannot get it to work but I think you probably can! Maybe an older version of GuiMods would get me up and running. 

Here are some screenshots from my system, which is a Raspberry Pi 3 running Venus OS version 3.12. 

[image]

[image]

[image]

Thanks for your help!

kwindrem, 2023-12-13:
I have found the problem and will have a new version out later today. Thanks for reporting.

lfhoward, 2023-12-13:
Thanks Kevin. Have a good evening.

kwindrem, 2023-12-13:
Should be fixed now: v8.26

closed 2023-12-13 by kwindrem

lfhoward, 2023-12-13:
> Should be fixed now: v8.26

Yes, working perfectly and it even preserved my settings. Thanks again.