kwindrem/GuiMods #217: v8.27 -> No file set for 3.11

SergioRius, 2023-12-14:
I suppose it automatically updated to v8.27, and guimods has completely deactivated.
The packages screen shows the message on the topic.
I don't think Venus v3.11 is an obsolete version.

EntwicklungScuddy, 2023-12-14:
Same here. I downgraded to v8.24. Everything is fine again. The same happend on a v3.12machine.

kwindrem, 2023-12-14:
I am not seeing a problem with GuiMods v8.27 on Venus OS v3.11 or v3.12.
There was certainly a problem with v8.26 but v8.27 was the fix.

SergioRius, 2023-12-15:
> I am not seeing a problem with GuiMods v8.27 on Venus OS v3.11 or v3.12. There was certainly a problem with v8.26 but v8.27 was the fix.

I don't understand. I always had the problem with the v8.27. This version doesn't want to install:
[image]
[image]

thomasinaz41213, 2023-12-15:
I have had this happen a few times. Have you tried rebooting Venus? Usually fixes it for me. If it was a FileSets issue it would usually just display "no FileSets".

kwindrem, 2023-12-15:
What GX device are you running (Cerbo, etc)?

I can not duplicate the issue here. I will need log files to diagnose further.

If you have ssh access:

tail -50 /data/log/SetupHelper | tai64nlocal

Please zip logs before posting here.

Use the PackageManager Backup function to store logs on a USB stick. This creates logs.zip which you can post here.

You might try downloading and installing GuiMods again. Using PackageManager is fine.

You can also move on to Venus OS v3.12 which is the current release.

Freensman, 2023-12-20:
Hey all

I facing the same problem. Tried up and down with 3.13 and 3.10 and 3.20 and GM 8.28. 

same message on all tries: "no file set for 3.11" (3.13 etc.)
I uninstall setup helper, uninstall guiMods ect. 
Reinstall all, but GuiMods is still there.

No success. 

Attached the asked logfile:

tail -50 /data/log/SetupHelper | tai64nlocal

[no_file_set_log.txt](https://github.com/kwindrem/GuiMods/files/13726522/no_file_set_log.txt)

[image]

Thanks for help. Freens

kwindrem, 2023-12-20:
Freensman, your issue is that in fact a new version of Venus OS was just released and I need to update GuiMods. I'll do that now.

Freensman, 2023-12-21:
Thanks a lot for your fast answer and help. 

Space is still available. 

It seams I can not uninstall GuiMods from the Setup Helper:

[image]

Is it possible without problems to just delete the folder directly, and reinstall after?
Or is there a direct possibility to download via ssh again?
(sorry I'm not a coder, just have basic knowledge about it)

[image]

Thanks so much - Freens

kwindrem, 2023-12-21:
Looks like PackageManager is stuck for some reason or the downloading message would go away.

Reboot and see if you can download. 

If a package is uninstalled (and in our case GuiMods is uninstalled), you can delete the package folder but you should then perform the Initialize PackageManager (at the bottom of the main PackageManager menu).

You can download from the command line using wget. That procedure is in the SetupHelper ReadMe.

Freensman, 2023-12-21:
Thanks for this. 

I tried various things, finally the following worked.

-> uninstall and reinstall SetupHelper over ssh-> don't solve issue
-> unistall GuiMods via SetupHelper ->  don't solve issue
-> delete GuiMods folder over ssh -> not able to reinstall over SetupHelper
-> uninstall SetupHelper over ssh
-> delete SetupHelper Folder over ssh
-> reinstall SetupHelper over ssh
-> SetupHelper runing 
-> GuiMods running without any install or download.

Thanks so much for your help and very appreciated work.

closed 2023-12-21 by kwindrem

SergioRius, 2023-12-22:
Please @kwindrem , can you tell me what I have to do to make Guimods work again?
Thanks.

kwindrem, 2023-12-22:
SergioRius, without logs, there's only so much I can do. Here are some shotgunning steps you may or may not have tried yet:

Use PackageManager to download GuiMods again and try the install manually.

Uninstall amy modifications authored by others then pdate the Venus OS firmware: switch to the stored backup then use the online update to install the firmware again. Do not simply switch back to the existing stored version in the backups section.

If you get GuiMods back, uninstall it, then install one of the other modifications, then install GuiMods again. Repeat the GuiMods uninstall, install other modification, reinstall GuiMods for each of the packages you are using. This will identify any conflicts between modifications.

Note that it is not necessary to do these modification steps with my packages, just those authored by others.

I will need detailed explanations or better yet the log files from a failed GuiMods install.

You can put the logs on a USB stick from using the PackageManager Backup function. It generates a logs.zip file on the USB stick you can post here.