kwindrem/GuiMods #218: Request for battery action buttons

sarah1331sarah, 2023-12-21:
Would it be possible to add buttons inside the battery detail screen 
[image]

battery selection buttons 
“Force discharging off” and “force charging off”
[image]

Am not sure if your going to discontinue this project 

thanks for reading

kwindrem, 2023-12-21:
There really isn't space in the battery detail to add buttons. I also don't see these parameters anywhere in the stock Victron menu code. Is this a third-party BMS?

As I have said elsewhere, GuiMods is going to be phased out when gui-v2 is released and GUI v1 is no longer available. It is not yet clear if or how gui-v2 can be modified but minds better than my own are working on how this might be possible.

sarah1331sarah, 2023-12-21:
It’s in https://github.com/Louisvdw/dbus-serialbattery
The driver but I understand

sarah1331sarah, 2023-12-29:
I don’t want to sound a complete a-hole bud but is this something you could do or is it impossible? I will try to figure it out but am struggling it would save me running wires are relays etc

kwindrem, 2023-12-29:
Not really possible. Plus with the new GUI coming I'm not all that motivated to makeing changes to GUI v1 overview and detail pages.

sarah1331sarah, 2023-12-29:
I totally understand thanks again

closed 2023-12-29 by sarah1331sarah