kwindrem/GuiMods #219: Trying to reach kwindrem

cuisinartoh, 2024-01-04:
I'm sorry to use GitHub to reach out, but I couldn't find a way to message or email you directly. I own a Victron distributor business and have demo systems set up and can give you access to a Cerbo GX or Ekrano GX (or both) remotely. I always have a split-phase system running in the shop and occasionally have three-phase systems setup before they go out the door. If you would like to have a system you can use for testing releases with multiple inverters (either parallel, multi-phase, or both) please let me know. It would be my small way of contributing to your projects. I can be reached at ejones@offgridps.com. Please send me an email if you're interested. I don't know my way around GitHub :-) 

Thanks

kwindrem, 2024-01-04:
Thanks for the offer. That will really help. I will contact you by email for system access.

cuisinartoh, 2024-01-04:
My pleasure. I'm a bit surprised VE hasn't set you up with a remote system. Maybe you haven't requested it. I'll be on the lookout for your email. If I don't see it come through in a day or so I'll message you back here in case it got lost along the way. Thanks.

kwindrem, 2024-01-04:
I have a small system in my travel trailer: one Multi Compact, an MPPT, a couple of SmartShunts and a Cerbo. It's sufficient for most of my debugging. I've relied on others for parallel and multi-phase testing or for testing on Quattros or systems with a managed battery.

I had plans to add a split-phase Quattro system to my house but the necessary UL and other certifications are missing and I need more than what a Multi can provide (~ 8 KVA per leg). Yes, I could parallel up Multis but don't really have the space.

So I appreciate the offer you've made.

cuisinartoh, 2024-01-04:
I get it. In a little over a week I'm attending a distributor meeting in San Diego and the topic of Victron's strategy for stationary systems in North America is at the top of my list. The Multi-II 48/3000 and 48/5000 have the UL certifications, including 1741 (SA? SB?) so it should pass inspection. However, VE hasn't released grid codes for grid interaction. They've been saying "it's coming" for years now and it's getting very irritating that it hasn't happened. 

There are some additional models coming this year, including a Quattro II, but I don't have information about what, if any, UL certs they'll have. And I don't yet know anything about legally feeding back to the grid with legit grid codes. 

Sorry, I guess I needed to get that out of my system. hehe.

I currently have three Pytes managed batteries connected to a system with two 48/5000 in split-phase. Today I just received two PowerSync 48V 100Ah managed batteries and I'll be putting them into the system so I can get some experience with their BMS. 

I also have training systems that each have two Multi 12/3000 in split-phase, Victron Smart Lithium batteries, and a variety of Victron smart BMS's. 

I can't promise that I can quickly reconfigure a system for you to test because I stay pretty busy. But, with just a bit of heads up I can probably get it done in a day or two. Let me know what you need.

jsalbre, 2024-01-04:
I know where getting off topic here, but they still have ground leakage on the Multi II that doesn’t meet NEC, so I’m not sure how they got UL certification.

cuisinartoh, 2024-01-04:
I'm not aware of ground leakage issues with the Multi-II 120V units. Do you
have any links to share re: that? Anyway, check this link out:
https://www.victronenergy.com/upload/documents/Certificate-UL-1741-and-CSA-C22.2-MultiPlus-II-48V-3kVA-120V.pdf

I misspoke about the 48/5000. I was sure it also had 1741 but it's pending
and I heard it was due to trying to get 1741-SA. I do NOT work for Victron
so take anything I say with a grain of salt. :-)

jsalbre, 2024-01-04:
Here’s a thread from the community all about the issue.https://community.victronenergy.com/questions/161020/multiplus-ii-2x120v-tripping-gfci.htmlJeremy

https://www.victronenergy.com/upload/documents/Certificate-UL-1741-and-CSA-C22.2-MultiPlus-II-48V-3kVA-120V.pdf

cuisinartoh, 2024-01-04:
Gotcha. That is the 2x120 model and not the entire Multi-II product line.
I'll admit that I wasn't aware of this problem because with all of the
Multi-II 2x-120 we have installed, only one of our customers ever reported
an issue charging with a 15A/20A GFCI and since no one else had ever
reported an issue I chalked it up as a local electrical issue. I'll have to
test this in my shop.

That said, the inverter you are talking about is definitely not intended
for use in stationary installations, let alone UL1741 for grid-tie.

jsalbre, 2024-01-05:
I was under the impression that others would have the same issue, having been designed to the same spec. I expect the leakage goes unnoticed though, as most models will never be in behind a GFCI.Anyway, sorry all for the very-not-GuiMods-related conversation. We now return you to your scheduled programming.Jeremy

cuisinartoh, 2024-01-05:
I think you hit it on the head. My customer base - and us here in the shop - haven't connected the AC input of one of those into a GFCI outlet. I'll try it in my shop. I'm going to also try this with other Multis and Quattros and see if it's an issue with any of those. As we focus our business more and more on larger systems and less on van and RV business, we just haven't experienced the issue when someone is trying to charge on a GFCI circuit. Thanks for the conversation!

cuisinartoh, 2024-01-26:
Hi Kevin. I didn't receive an email from you. The offer still stands. I would also like to chat with you sometime about assisting with development of a driver in Venus OS for a product we're building. That could come in the form of trade or payment. Please get in touch when you have a moment.

kwindrem, 2024-01-26:
I'm probably not the best person to help with drivers. I haven't looked into that part of the system and there are may others out there developing drivers for their own devices, mainly BMS.

cuisinartoh, 2024-01-26:
No worries. I have a couple of others I can discuss it with, including one who recently wrote a driver for their battery/bms. I thought I'd start with you :-) Thanks.