kwindrem/GuiMods #222: Water generator not showing on overview page

Pearsonhenry2, 2024-01-21:
We have a smart shunt setup to monitor input from a water generator (Watt & Sea unit).
On the overview page it is not showing up. If I change it in the app to type "wind generator" it does show.
Any way to allow water generators to show?
[image]
[image]

kwindrem, 2024-01-21:
Victron added support for Wind but not Water in the GX device. Water generator could be added to system calc and a new tile created on the overview but I'd need to prioritize which is shown if there are more than will fit on the page.

Also, I have announced a pause in GuiMods development due to the new gui-v2 from Victron that will address many of these issues.

Pearsonhenry2, 2024-01-21:
Thanks for replying so fast!
For now we’ve just changed the shunt over to a wind generator and will just accept it’s the wrong label.

We don’t have a Victron inverter so this overview page is much better than the standard victron gui for us.