kwindrem/GuiMods #224: Digital inputs missing

thomasinaz41213, 2024-01-25:
Guimods v8.31 and Venus beta v3.20~40
Digital inputs are listed under settings/IO/Digital inputs but they are missing from guimods overview page. Tanks and temps are there, but where the digital inputs usually are it just shows "no digital inputs". Again, just to clarify, if I go into the settings menu and IO the digital inputs are listed with the correct type selected.

I am aware there was an issue with a previous beta and digital inputs disappearing so not sure if this is part of that issue. This was updating from v3.14 to beta v3.20~40. going back to v3.14 the inputs reappear in guimods overview.

kwindrem, 2024-01-25:
I can not reproduce this issue. I would need logs to troubleshoot further.

thomasinaz41213, 2024-01-26:
Went back to v3.14, uninstalled guimods, fresh installed v3.20~40, then installed guimods and digital inputs is back. I dunno. Not the first time strange things have happened. Several reboots didn't help, but another fresh install seems to have done the trick. Sorry to have wasted your time.

closed 2024-01-26 by thomasinaz41213