kwindrem/GuiMods #225: my display stopped dimming (turning off)

ondrom111, 2024-01-28:
After installing GuiMods9.0 and VeCan 3.9, my display stopped dimming (turning off).

kwindrem, 2024-01-28:
Neither package has anything to do with display blanking or dimming in the case of GX Touch or RPI touchscr

Please verify that uninstalling these packages restore the screen saver. If not, the problem lies elsewhere. 

Also make sure screen saver is not set to off.

If you can isolate the cause to one specific of my packages, please send me the logs. You can do this with the backup function in Package Manager. That will create a logs.zip file on removable media (or in /data which you can then transfer to a computer via scp, etc.) Then post here.

ondrom111, 2024-01-28:
[logs.zip](https://github.com/kwindrem/GuiMods/files/14077817/logs.zip)

Thanks for help

kwindrem, 2024-01-28:
What version of firmware are you running?

ondrom111, 2024-01-28:
3.14

kwindrem, 2024-01-28:
I can not reproduce the problem here.

What platform? (Cerbo, CCGX, RPI, etc)
What display?