kwindrem/GuiMods #226: No FileSets Venus beta V3.20-45

thomasinaz41213, 2024-02-04:
No FileSets for Venus os v3 20-45

kwindrem, 2024-02-04:
Testing now.

BTW, ExtTransferSwitch is being merged into GuiMods due to all the overlap in modified files.

Zeoic, 2024-02-04:
I am seeing the same message on my pi

kwindrem, 2024-02-04:
Please be patient. I'm working on the situation. Many changes that all need to be checked across multiple firmware versions. Will release tomorrow.

Zeoic, 2024-02-04:
> Please be patient. I'm working on the situation. Many changes that all need to be checked across multiple firmware versions. Will release tomorrow.

Thanks for looking into it! This looks like a super handy package. Appreciate all the work you put into all these!

kwindrem, 2024-02-04:
Now fixed

GuiMods now includes the functionality of ExtTransferSwitch package. ExtTransferSwitch will automatically be removed when installing the new GuiMods.

closed 2024-02-04 by kwindrem