kwindrem/GuiMods #230: GuiMods downloading but not installing

JohnnyOckerman, 2024-02-15:
Hi,
I’ve successfully installed the setup helper and see Package Manager at the bottom of the Settings menu. 
I can add GuiMods from inactive to active packages.  However, after downloading the GuiMods package the Install button doesn’t become accessible.  The version in the GitHub field is visible but not in the Stored field.
Any advice to get the GuiMods installed?
Thank you,

kwindrem, 2024-02-15:
The blank "stored" version indicates GuiMods is not on the system. You need to download if from GitHub. You can do this by turning on automatic downloads and then automatic install to get it to actually install. Or you can tap on GuiMods in the active package list and do the operations manually. Once downloaded, the install button should become active.

JohnnyOckerman, 2024-02-15:
Hi, thx a lot for the swift reply.  Unfortunately, the automatic nor manual download is successful.  When clicking Proceed with manual download It says “… downloading” but Stored field an Install button remain empty/inaccessible.

kwindrem, 2024-02-15:
Try a reboot if you have not done so already.

Which SetupHelper version is installed?

Is there anything in your internet connection that could prevent the download?

I'd need to take a look at the logs.

Use the backup function inside PackageManager. That will create a logs.zip file on removable media which you can post here.

JohnnyOckerman, 2024-02-16:
No success after multiple reboots.
No idea about an internet issue.
Cerbo GX Firmware  version: v3.20
SetupHelper version: v6.6
logs.zip attached.
[logs.zip](https://github.com/kwindrem/GuiMods/files/14310255/logs.zip)

JohnnyOckerman, 2024-02-18:
Hello,
Is the logs.zip helpful to identify the issue?
Under package manager I read something like file(s) not available. 
Thank you,

JohnnyOckerman, 2024-02-21:
Hello,
I did reset my Cerbo GX to factory settings and did reinstall SetupHelper v6.6.
Unfortunately the issue remains unresolved.  After downloading the GuiMods, I see a brief message "downloading GuiMods..." but the Install button remains greyed out.
Thank you,

kwindrem, 2024-02-21:
SetupHelper v6.6 has a problem that prevents downloads if there is no package already stored on the GX device.

SetupHelpver v6.8 is the current version and fixes this issue. You need to upgrade SetupHelper.

closed 2024-02-21 by kwindrem

JohnnyOckerman, 2024-02-21:
Awesome, your advice settled the issue. 
Thanks so much.