kwindrem/GuiMods #231: GuiMods fails to install after VenusOS 2.30 Update

petertmac, 2024-02-16:
I updated VenusOS on my Raspberry Pi 3B+ today and after a reboot I lost GuiMods. Looking at setup helper it shows a message under GuiMods as “no file set for v3.20”. I tried updating to v10.3 of GuiMods and same error. I rolled back to VenusOS v3.14 which was working, however it updated to v10.3 and I have the same error. Any suggestions on how to recover?

albundy0815, 2024-02-16:
great, that I'm not the only one. I was wondering if it is a problem on my side. Because just before I've updated to 3.20

petertmac, 2024-02-16:
Correction- typo on my part. The upgrade was to 3.20.

ckvsoft, 2024-02-16:
It's probably a general problem.
I have the same problem here.

Bosode, 2024-02-16:
yeah - same at my installation. early this morning fallback to Victron default layout and menu structure.

dersch81, 2024-02-16:
yes same here

kwindrem, 2024-02-16:
Sorry for the install failure. Somehow the version I checked in had a corrupted file set for v3.20.

V10.4 is being pushed as I write this

closed 2024-02-16 by kwindrem

pcmedicsd, 2024-02-16:
Yahoo! thanks for the quick fix Kevin :)

albundy0815, 2024-02-16:
@kwindrem thank you so much for support and also for the GUIMods. Today I've seen, that the default VenusOS is ~1% and your addon gives me additional 99% at user experience ;-)

petertmac, 2024-02-16:
Amazing, woke up this morning and the fix is done. Thanks Kevin.

Bosode, 2024-02-17:
@kwindrem Thank you for your amazing job to build an update this great tools and share them with us ***thumbs up**