kwindrem/GuiMods #233: VRM Consumption Data Wrong After GUI Mods install

atrane22, 2024-02-18:
I installed GUI Mods on my cerbo GX on february 12th. Since then, when i check my consumption data on my VRM dashboard it only shows consumption when the inverter is turned on and doesn't show anything else consuming power. Before GUI Mods install on the 12th it showed complete data but every day since then, only shows inverter consumption.

I have 2 screenshots attached. One shows consumption tracking before GUI mods, the other shows the way it looks now after GUI mods. During both charts, our inverter was turned off over night. 

Is there a reason for this? How do i fix it?
[image]
[image]

kwindrem, 2024-02-18:
I checked my system and the internal consumption values that VRM uses to create these graphs behave the same with and without GuiMods installed: When the Multi is off, neither AC consumption power and AC consumption power on output is reported. In all other modes, AC consumption power values are valid. 

I am looking at internal AC consumption numbers and do not see any DC consumption numbers except of DC System which shows the same numbers with GuiMods installed or not.

So I am at a loss as to why one graph is showing consumption and the other is not.

atrane22, 2024-02-19:
That's really odd. I've always had my consumption chart show full consumption in a 24 hour period and we almost always turn our inverter off at night or when it's not needed. Like i said, i was able to track down exactly when it switched and it was on the day i installed GUI Mods. 
I'm not worried about AC consumption data just the total consumption of my system that shows on my VRM dashboard. Even when i choose "consumption" on the dashboard it doesn't show consumption when inverter is off but this should show all DC consumption. 

[image]

kwindrem, 2024-02-19:
Can you uninstall GuiMods and run for a couple of days to confirm it's something in GuiMods. GuiMods does make some changes in systemcalc that VRM uses for accumulating consumption but as I said, my system is not seeing any changes to the data with and without GuiMods.

For your DC consumption, are you using a shunt to provide a real measurement? If not, systemcalc calculates this value. My system does have a shunt which might be the difference. Let me know and I'll test my system without the shunt.

atrane22, 2024-02-19:
I can give that a try this week when i get some free time. 

Yes, we use the victron smart shunt for our measurements.

kwindrem, 2024-02-19:
Just so I know for sure: you are using a smart shunt set to DC Meter - DC System?