kwindrem/GuiMods #234: DC power consumption inaccurate with GuiMods

schl0r, 2024-02-23:
Hi, I was looking forward to giving GuiMods a go on my new CerboGX in my van install, after everything I heard. Installed the latest version 10.4 today through the PackageManager.
Weird thing is that the DC consumption (lower middle in the screen) now is inaccurate, e.g. when having a consumption of 69W according to the SmartShunt, DC power only shows 23W. Multiplus inverter is turned off, PV does not generate any power, so there are no other electricity consumers/producers other than a few lights in the van.
I uninstalled guimods and the value was accurate and matched with the  battery wattage again. Would love to use GuiMods but not with inaccurate numbers. Anything that I missed?
Attached two images, one with guimods and wrong figures, one without guimods and correct numbers.
[image]
[image]

schl0r, 2024-03-01:
I think I got it, was running an older firmware on the Cerbo (2.91) and just updated to 3.22, also updated GuiMods to 10.6. Now DC Power is accurate, Although a second delay compared to the battery power, but that's a trifle.

kwindrem, 2024-03-01:
There is a difference in the calculated DC power between v2.91 and 3.22. A non VE.Bus inverter is ADDED in the v2.91 and SUBTRACTED in the v3.22. But the code is the same for the stock and GuiMods calculation. It's possible the sign of one of the contributors to the calculation is different in other sections of the code or even in the driver for the device. So I'm at a loss to explain the difference. But glad things are fixed for you now.

GuiMods averages calculated DC power over three seconds because the calculation often jumps around wehvarious loads and sources are changing. The averaging doesn't eliminate the jumping but does lessen it. The side effect is the value displayed is delayed 3 seconds compared to the stock code.

schl0r, 2024-03-01:
> GuiMods averages calculated DC power over three seconds because the calculation often jumps around wehvarious loads and sources are changing. The averaging doesn't eliminate the jumping but does lessen it. The side effect is the value displayed is delayed 3 seconds compared to the stock code.

Thanks for the explanation.

closed 2024-03-01 by schl0r