kwindrem/GuiMods #238: Generic I/O  type missing Digital Inputs page

Lucifer06, 2024-03-04:
Type Generic I/O ( type=10 ) is not supported in PageSettingsIo.qml (cf /opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py).
This is currently listed as "Unknown" while on the main page this is correctly tagged as  " Generic I/O "
There is also a limit in the PageSettingsIo.qml that displays only the first 9x Digital Inputs

kwindrem, 2024-03-04:
The type shown on in the device list comes from dbus_digitalinputs.py while the type shown on the I/O page comes from MbItemDigitalInput.qml. GuiMods replaces this file however the stock file also does not include Generic I/O.

I can certinly add Generic I/O to GuiMods but the question of why it is not included should probably be directed to Victron. Doing that now.

Lucifer06, 2024-03-04:
Makes sense

closed 2024-03-20 by Lucifer06