kwindrem/GuiMods #242: Guimods missing in Venus 3.30

drtinaz, 2024-03-19:
Updated to firmware 3.30 and guimods shows as installed in the active packages list, but is missing from display & language menu. Gui is oem. Gui restart or system reboot does not remedy the problem.

Setuphelper 7.6
Guimods 10.15

kwindrem, 2024-03-19:
My bad. I was checking remove on the last package in the list. Removing one in the middle does corrupt the list because PackageManager is crashing and restarting in the middle of the operation. v7.7 coming soon.

blieven1, 2024-03-20:
I'm having the same issue

Used Setuphelper 7.7 to install Guimods 10.15

[image]

Shutdownmonitor is available in the menu but guimods not

Kind regards,
B

kwindrem, 2024-03-20:
Noted. Working on the problem now. 

There's a bug with automatic downloads that causes PackageManager to crash. For now, please turn off auto downloads.

blieven1, 2024-03-20:
Hi, I checked but automatic downloads are already off. Tried uninstalling and redownloading but nothing helps

drtinaz, 2024-03-20:
When this happened to me yesterday I was able to get guimods installed by reverting to v6.10 of setuphelper and rebooting twice. Then went back to latest v.

kwindrem, 2024-03-20:
SetupHelper v7.8 should be the fix. I would recommend downloading and installing GuiMods and ShutdownMonitor manually to make sure everything is installed properly.

blieven1, 2024-03-21:
Thanx! It works now

kocki2, 2024-03-22:
It's working again:
GuiMods 10.17
SetupHelper 7.9

Thanks,
Gerd

closed 2024-03-22 by kwindrem

Albarge, 2024-03-22:
I just updated SetupHelper from v7.8 to v7.9 and GuiMods from v10.16 to v10.17 then VenusOS from v3.22 to v3.30 and had the same problem, both BatteryAggregator and GuiMods were uninstalled like before after the Venus update.

[image]

kwindrem, 2024-03-22:
I have tried several times to reproduce this problem but so far the automatic install is working for me.

I would appreciate the logs from your system:

Insert a USB stick into the GX device, go to Settings / Package manager / Backup & restore
Choose Backup settings, logos, logs ( not to local storage). This will create among other things a logs.zip file on the USB stick. Please post logs.zip here.

Albarge, 2024-03-22:
Thanks for looking, much appreciated!

Not sure if relevant, but as I mentioned on the Victron forum, it turned out dbus-serialbattery was actually broken by VenusOS v3.30 also, I found reverting back to an older serialbattery worked. There was a mention of missing files in the install command window for the older version of serialbattery driver after it installed.

I'll just get the Package Manager logs now

CaptainRivet, 2024-03-24:
Hi Kevin,
Had updated to 3.30 and with 7.7 setup helper and v10.15 was not working. Directly updating to 7.10 and v10.17 gui mods just now and still not working. Did manual update. There was no reboot after the install but I did one manually and no change…gui mods still not working. Have a cerbo GX.

kwindrem, 2024-03-24:
CaptainRivet, I need lots more information that "not working". Any messages displayed on the GUI? What's shown for the three versions?

Also, logs would be very helpful. See instructions above.

CaptainRivet, 2024-03-24:
No message shown. Screenshot of the p
ack manager with versions see blow.
[image]
Gui mods simply not working, setting is on the bottom of the 
Page.
[image]
[image]
[image]
Rebooted cerbo, no change. Venus V3.30 Large.

CaptainRivet, 2024-03-24:
How can i re-install the V10.17 again?
There should be a automatic reboot after i installed 10.17 from V10.15? There wasn't.
Every update before it did but not this one.

CaptainRivet, 2024-03-25:
Figured it out. 
Downgrading to V 10.16 gui mods and then upgrading to 10.17 didn’t work. Uninstalling gui mods and reinstalling 10.17 also didn’t work as gui mod is again not asking for gui restart during install.
What worked was uninstalling setup helper and gui mods and then install setup helper 7.10 via USB stick and then 10.17 gui mods, gui mods now working.
It seems you cannot go 7.7 to 7.10 setup helper and then install 10.17 gui mods, something is missing as gui mods is not asking for gui restart. From scratch works.

kwindrem, 2024-03-25:
Sorry, can't help without logs.

kwindrem, 2024-03-25:
You can always download again from the Package editor menu.

If you want a specific older version, set the branch/tag to that version then download it.

CaptainRivet, 2024-03-25:
Kevin I figured out and gui mods working now. Described what I had to do in case someone has the same issue then I do.