kwindrem/GuiMods #243: VenusOS v3.30 support issue

Lucifer06, 2024-03-20:
Trying to install GuiMods v10.15 from clean VenusOS v3.30
Setup was conducted from the GUI. No errors. Restart of the GUI was requested, and executed.
Then SetupHelper is still showing GuiMods not installed, and effectively it is not installed in the GUI display Menu.
There is an error reported in the logs: no soure file for DcSystemRow.qml

2024-03-20 15:03:32.968185500 GuiMods/setup: --- starting setup script v10.15 action CHECK
2024-03-20 15:03:33.010193500 GuiMods/setup: creating package options directory /data/setupOptions/GuiMods
2024-03-20 15:04:18.012248500 GuiMods/setup: --- starting setup script v10.15 action INSTALL
2024-03-20 15:04:18.286529500 GuiMods/setup: installing files
2024-03-20 15:04:18.592971500 GuiMods/setup: ERROR: no soure file for DcSystemRow.qml
2024-03-20 15:04:18.637747500 GuiMods/setup: installing services
2024-03-20 15:04:18.662745500 GuiMods/setup: INSTALL failed - attempting UNINSTALL in endScript
2024-03-20 15:04:18.720776500 GuiMods/setup: uninstalling files
2024-03-20 15:04:18.996651500 GuiMods/setup: restarting generator service
2024-03-20 15:04:19.025522500 GuiMods/setup: restarting systemcalc service
2024-03-20 15:04:19.048457500 GuiMods/setup: restarting digital inputs service
2024-03-20 15:04:19.090223500 GuiMods/setup: completed - GUI restart needed

Lucifer06, 2024-03-20:
Additional info: SetupHelper is with last versions v7.7

A manual install from the shell reports the same error but the difference is that now PackageManager reports GuiMods to be installed while this is not the case as I don't see the Dimply menu altered.

root@einstein:/data/GuiMods# ./setup 
--- starting setup script v10.15 action NONE

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i
installing files
ERROR: no soure file for DcSystemRow.qml
installing services
restarting generator service
restarting systemcalc service
restarting digital inputs service
Restart the GUI now (y) or issue a do it manually later (n): y
completed - restarting GUI

kwindrem, 2024-03-20:
Working on it.

Ritterlanzelot, 2024-03-20:
Good Evening, on trying to update, I get "incomplete file set for v3.30". Do you have any recommendment for a solution?

drtinaz, 2024-03-20:
There is an issue with setuphelper that Kevin is currently working on. I had the same issue which I was able to get around by setting the setuphelper to v6.10 and then installing guimods again. If you do that and still get the fileset error, then switch the GX to the backup firmware and download/install a fresh copy of v3.30 and try again. I saw this once AFTER Kevin had already pushed the guimods update for 3.30. My guess is setuphelper was not able to complete something so a fresh firmware was required to 'reset'.

kwindrem, 2024-03-20:
OK, give SetupHelper v7.8 a try. 

I had some rather nasty thread interactions that caused PackageManage to crash while download and installing packages. I did not see those thread interactions on my Raspberry PI test system but did on my Cerbo which has a slower CPU than the PI.

I also found a bug that caused ALL packages to download after PackageManager started. That's fixed also.

The combination of those issues resulted in an incomplete package directory and/or partially installed packages. If you have seen errors like those reported above, try manually downloading and installing those packages after updating SetupHelper to v7.8.

Ritterlanzelot, 2024-03-20:
Works perfect - solved! Thank you so much for your work !!!

Lucifer06, 2024-03-20:
Good job, I'm back to business :-)
Had to clean a bit all the installs before : manually removed all the packages, including SetupHelper, deleted all the Packages from /data and erased dbus /Settings/PackageManager

closed 2024-03-20 by Lucifer06