kwindrem/GuiMods #244: Missing PV Inverter names

blieven1, 2024-03-21:
Guimods 10.16

At the moment it displays -- instead of the name
[image]

They are named in the device list:
[image]

The inverters are in fact the new Victron VM-3P75CT energy meters in PV role, connected via ve.can bus:
[image]

If I can help debugging let me know.

Kind regards,
Bjorn

kwindrem, 2024-03-21:
This is likely because you are using an energy meter to supply the information for your PV inverters. a /CustomName parameter is needed in the dbus service for them. 

Can you send me screen shots one one of the devices in dbus-spy?

dbus-spy is a command line tool for inspecting dbus services.

blieven1, 2024-03-21:
Sure, here you go: (it does seem to be filled in)

[image]

kwindrem, 2024-03-21:
Very interesting. I'm getting the name from the same dbus service as the power and as you show above the name is present.

Very odd in fact since I show a single dash "-" if the name field is not valid, not two dashes "--".

What version of firmware are you running so I can check the specific code for that version?

What does PV Inverter Details page show? (tap on the PV Inverter tile.)

In dbus-spy while showing the page you sent above type the 't' key to toggle between value and text attributes. If the name changes to '--' then I know what the issue is.

Thanks

blieven1, 2024-03-21:
Hi,

I'm on v3.30

with 't' its indeed empty:
[image]

PV details is correct:
[image]

kwindrem, 2024-03-21:
I'll change my code to use .value instead of .text which is the difference between the overview and detail.

Thanks for reporting.

Are you using stock Victron drivers for the device or something a third party wrote?

If stock Victron I'll let them know there's a problem.

kwindrem, 2024-03-21:
Just released v10.17. Please check ASAP as I have no way of testing here. Thanks.

blieven1, 2024-03-21:
Yes,

Working now!

I used indeed all standard victron meters, no special drivers

Thanx for fixing!

kwindrem, 2024-03-21:
Thanks. I reported to the Victron forum

closed 2024-03-21 by kwindrem