kwindrem/GuiMods #245: Smart Shunt Configured as AC Charger Does not show up correctly

ecriscuolo, 2024-03-26:
I have a sailboat system with NO Multiplus, or any other inverter/charger.  Instead, a simple AC powered shore charger is hooked up through a Victron Smart Shunt to measure the charging current.
The Smart Shunt in configured as an "AC Charger" using the Victron Connect App.

In the GUI Mods flow display, it gets lumped in with the DC system instead of showing up as an "AC Charger" block.

kwindrem, 2024-03-26:
It is true that any charger power is included in DC system power. The stock Victron calculations included it so GuiMods included it also. The issue is that the stock Victron flow and the GuiMods Simple flow pages do not show charger blocks, only the GuiMods Dc Coupled and Ac Coupled flows do and only on AcCoupled if there is space.

In your system, there would be space for an AC Charger tile but that is true only for systems without a Ac input tile.

You could add a shunt for DC System power and that would then show in the DC System tile, but you'd still be missing the larger tile. 

I do show a tile for alternator in place of the AC input tile and could do the same for chargers but it would be one or the other.

kwindrem, 2024-03-26:
FYI, I believe work is being done so that gui-v2 WILL accommodate alternators and AC chargers and to show a more accurate calculated DC Loads power. I have not seen any changes to the DC System calculations in systemcalc so my guess is this is all handled in gui-v2. GUI v1 is the only place that uses the DC system value calculated in systemcalc.

I have no plans to make a GuiMods for gui-v2. It would be a complete rewrite and most of the enhancements in GuiMods are part of gui-v2: more tiles shown on the overview page, drill-downs to show details (like GuiMods Details pages).