kwindrem/GuiMods #247: changing display name in PV-inverters tile in panel over overview

knirma, 2024-04-18:
Hello,

where are these names generated in the PV Inverter tile in panel overview ?
Where can i change it 

[image]

I want to change the " - " into a real name.

Many Thanks

kwindrem, 2024-04-18:
Each device in the device list has a Device submenu. The name (if there is one for that device) is in there.

knirma, 2024-04-18:
Hello,

in this case it doens't seems to be.
Maybe a issue for the pico script owner. I tried it.
But he doesn't respond anymore.

[image]

[image]

[image]

Maybe you can help me.

Regards

knirma, 2024-04-18:
The opendtu will report the device names correctly.

kwindrem, 2024-04-18:
If the author of the driver for your device doesn't provide the /CustomName dbus parameter, then there's not much GuiMods can do to display a name.

knirma, 2024-04-18:
ok
thanks for the info

knirma, 2024-04-19:
i had found it because of your hint with /Customname
It was in the file /data/venus_kostal_picoTkostal_inverter.py

the original sequence was 
#self.dbusservice.add_path('/CustomName', "PLC Mec meter")
Why here the " - " was shown , i dont know.
Now it is "Piko55"  , as i like it.

Many thanks for your help !!

Regards