kwindrem/GuiMods #249: Cerbo GX (with Guthubmods v7.18) shows battery at 0.0%

Spartan61, 2024-04-28:
I’ve installed Githubmods v7.18 to my Victron Multiplus 12/2000/80-50 120V (400aH lifepo4 battery bank, 700w solar) + CerboGX display system but am getting a display that shows 0% battery charge even though when disconnected from shore power the inverter is supplying power (obviously the batteries have available power, enough to run the air conditioner, etc.).  What might I do to correct or troubleshoot this display anomaly? I put the system (via Cerbo GX remote) to bulk/absorption mode thinking it was something in my settings but still no apparent correlation between that mode and displayed uptake of charge to the battery bank. 

[image]

[image]

When viewing the status of my Multiplus 2K inverter/charger within the “device list”, it shows “state” as “equalize”, “input current limit” at 29.0 (I’m attached to a 30amp shore power source, “state of charge” as “- -“. Perhaps I have some setting “set” incorrectly . . . Or there is some component of the Githubmods I failed to add (correctly or at all)?

[image]

Any assistance or insights would be greatly appreciated!

kwindrem, 2024-04-28:
GuiMods gets battery state of charge from the values calculated by stock Victron code. This comes from the shunt connected to the battery or from the one inside the Multi. If you uninstall GuiMods you should see the same SOC value. Since you have a DC system, the battery monitor inside the Multi will not provide accurate information since it can't incorporate power the flows out of the battery into your DC loads or from your PV charger.

There are a couple of things that can cause the SOC to read zero. The battery monitor/shunt needs to see the battery reach absorption voltage with little current for a period of time. All these parameters are adjustable for the shunt but I'm not sure about the internal monitor since I've never used it. These values will depend on the type of battery. If these conditions are not met, the battery monitor can't calculate SOC. The battery monitor resets its SOC when turned off and then requires the above process to reset it's SOC.

I notice the system reports your battery is in equalize mode. This should be disabled as it could harm lithium batteries. Equalize if for lead-acid batteries only.

Spartan61, 2024-04-28:
Thanks so much for the quick reply! 

1st, yes, I somehow set the system to “equalize” but changed that pretty much immediately after reading the same comment on the Victron website. Hoping that the short amount of time it was “on” (to “equalize”) didn’t do any harm! Will monitor the system for issues. 

2nd, while I THINK I understand your explanation regards the SOC readings vis-a-vis the internal battery monitor within the Multiplus, I wanted to also point out the system was installed or includes both a Victron BMV-712 Battery Monitor and a Victron VE.Bus Smart Dongle (not the V2 version but the earlier smaller VE.Bus). Would either of these (singularly or in combination) not allow the GuiMod “see” and report the SOC?

Lastly, if there is no way to see and monitor SOC (if there is, I’d really be interested in hearing how I could achieve it), what choice/choices do I have, if any? I really love the GuiMod work you’ve done and it’s so much better than the stock Victron Cerbo GX gui! I’d hate to have to revert back to the original system.

Again, I’d be immensely thankful for any and all advice your able to provide on this topic.

V/r,
Spartan61

Spartan61, 2024-04-28:
Thanks for the advice and quick reply.

If I might ask a couple of more questions. First, the system has both a Victron BMV – 712 Battery monitor as well as the Victron VE.Bus Smart Dongle installed. Is there anyway using these devices to relay battery SOC to the Cerbo GX running GuiMod?  (Btw, not that it changes anything but earlier i incorrectly stated that I have four Battleborn LiFe2PO4 batteries when I actually have five!)

Second, if it’s not possible to configure the system (with its current configuration and components) to show SOC (something I’d really want to have a capability for as we often go out boondocking - our system is installed in our 2021 NorthernLite 8-11EX/LE truck camper), is or are there ANY components that I can add, or modifications I can make, that would facilitate the goal of tracking and seeing SOC?

kwindrem, 2024-04-28:
First, let's rule out problems with GuiMods. Please uninstall it and see if the stock Victron code does or not show SOC. I suspect that it will show the same as GuiMods. If this is a problem with GuiMods then I will need to look at the logs but I'll save that for later.

Make sure your BMV-712 is shown in the device list. If not, check its VE.Direct connection. You must connect all your devices to Cerbo via a cable. Cerbo can't read info over bluetooth!

By default, Cerbo will automatically select a battery monitor. But check to make sure that the BMV-712 is selected.

Victron typically ships their devices configures for lead-acid batteries. Check the setup of your BMV-712 and all your charging sources: Multi, PV Chargers and anything else that might be charging your house battery. These all need to be programmed to match your battery types. Check the manufacturer for actual values. There's a decent description of how to set up the BMV-712 in its manual.

One setting of note on the BMV-712: Reset SOC on power up should be off. Otherwise it will set the SOC to 100% if you disconnect power to it.

Run a short test with a heavy AC load like the water heater, microwave. Once the load is running, switch the inverter to invert only or disconnect shore power and watch the battery voltage. It should drop when inverter kicks in but should not continue to drop over time. A few minutes should be enough to verify your batteries are in fact charged and can handle your load.

Once you have all your devices programmed, watch the battery voltage and see if it ever reaches the set absorption voltage. If not, you may need to remove DC loads from your system until it is reached and SOC resets to 100%. The SOC reset may take some time. The battery must reach absorption voltage and stay above the value set in the BMV with battery below the tail current for the specified time.

Note: ExtTransferSwitch functionality is now part of GuiMods so you should remove the ExtTransferSwitch package. It won't do any harm if it's there and you won't be able to install it so this is just cosmetic.

Albarge, 2024-04-30:
> First, the system has both a Victron BMV – 712 Battery monitor as well as the Victron VE.Bus Smart Dongle installed. Is there anyway using these devices to relay battery SOC to the Cerbo GX running [...]

How are your Battleborn batteries communicating to Venus? If you are using Battery Aggregator to aggregate them, and it can see your batteries then you can set in that by editing the config.json file for what you want to be used for SOC, see the aggregator  ReadMe.

Albarge, 2024-04-30:
Maybe that's why SOC is not showing as Battery Aggregator doesn't have any batteries to aggregate, there's no reason to install it.