kwindrem/GuiMods #251: Where to set generator current limit?

janzwiegers, 2024-06-04:
I cannot see how to set the generator current limit like in the ExtrTransferSwitch?

kwindrem, 2024-06-04:
The input current limits for grid and generator are stored the external transfer switch changes between sources. Setting the input current limit as well as other parameters: (grid vs shore, stop when AC available) is done using the normal controls:

set the external transfer switch to the grid position
set the input current limit to the value for the grid and stop when AC available and input type (grid or shore).

set the external transfer switch to the generator position
set the input current limit to the value for the generator

Operate the system normally and the current limit will change when the external transfer switches sources.

Lets say the grid current limit is 50 amps and the generator current limit is 12 amps.

When the transfer switch changes to generator, the current limit is set to 12 amps

When the transfer switch changes to grid the current limit is set to 50 amps.

Now the input current limit is changed to 30 amps while on grid.

Switching to generator will set the limit to 12, then switching back to grid will set the limit to 30.

janzwiegers, 2024-06-04:
OK so if I understand this correctly the input current must be set when the external transfer switch is in that position?

Currently I can only see one specific place to set the current limit.

 

Thank you for the reply!

closed 2024-06-04 by kwindrem