kwindrem/GuiMods #253: Hide Inverter

kuhnboy, 2024-06-12:
I have a 48v/12v setup where I have all of my inverting and ac loads on the 48v side with a Cerbo gx. I’d love to either:
- See my 12v devices in the device list but from my understanding the 48v system would affect the 12v MPPT if I connected it with a ve direct cable. (I have a smart shunt as well on the 12v side)

or

- Setup a pi / Cerbo to show 12v devices separately but is there a way to hide all AC parts of the tiles / diagram with guimods?

kwindrem, 2024-06-12:
I'm not sure about #1.

Some tiles hide based on presence of the related component but others like the inverter do not.

kuhnboy, 2024-06-13:
@kwindrem do you have contribution guidelines and/or accept pull requests?
[image]

kwindrem, 2024-06-13:
I don't have any formal contribution guidelines but others do occasionally submit revisions via pull requests. Generally, they miss things like compatibility back to v2.7x and/or a version number change. So I've been manually incorporating their changes into my repo and rejecting the pull request.

I do have a document describing how to author a new package suitable for SetupHelper:

https://www.dropbox.com/scl/fo/bx5aftvgrqq0vp060mwip/ANKdSqoEP9MC06R-rUoIExQ?rlkey=razevzfts259w2v2r9ajeet2g&st=w2bkepz3&dl=0

It seems like you've been able to make the changes you need for your specific needs and I probably would not be inclined to incorporate them into Gui Mods.

The document I linked above describes how SetupHelper / PackageManager deals with reinstalling mods after a firmware update which might be useful to you. Or you could create a package for your personal use.

kuhnboy, 2024-06-13:
Thanks for the response. If you feel that a 'Hide AC System' toggle does not align with guimods, then I'm fine creating my own package. I know there have been some others around the community that have asked for this, so it's not unique, but may not be a great fit for guimods so I would defer to you on that guidance. Thanks!

kwindrem, 2024-06-13:
I made a few minor tweaks to the code and was able to hide the multi tile and the connections without any additional user preferences. I'll incorporate these changes in GuiMods

GuiMods simple flow overview
[image]

GuiMods DC & AC coupled overview
[image]

kwindrem, 2024-06-13:
v10.39 has these changes

closed 2024-06-13 by kwindrem

kuhnboy, 2024-06-13:
Excellent. I can't wait to try this out.