kwindrem/GuiMods #254: Version 10.39 removes inverter even if it is just turned off

Bucket1616, 2024-06-15:
After upgrading to v10.39, our inverter tile for our MultiPlus pair disappears after it is turned off. 
Given the standby current of the inverter pair, we routinely power off inverter pair off to save power. In prior versions, the tile would indicate “Off”, but still be present and clickable so that we could easily turn back on when needed.
With the tile now disappearing, turning the inverter on is nowhere near as easy as it was prior to this update. 
Can you please consider correcting traía so that it can remain present after being switched off, but never show up if an inverter was never present to begin with? Thanks!
P.S. Though not ideal, even having the software remember inverter presence the first time it is powered on would be acceptable. Meaning that upon rebooting with the inverter off, the tile would not show, but as soon as it is powered it would appear and not disappear. Just a thought in case it is hard to do otherwise.

kwindrem, 2024-06-15:
Thanks for reporting. This was not the behavior I had planned when someone asked if I could hide the inverter tile for systems that don't have one. I'll look into this.

Bucket1616, 2024-06-15:
Thank you!  I was just glad that when we checked under our boat's salon 
seats, the inverters were still there.  Wasn't sure how good your 
software was... ;-)

Mike

it is just turned off (Issue #254)

kwindrem, 2024-06-15:
v10.41 was just released that should fix this.

Note that there's a bug in Victron code that doesn't fully detect the presence of the multi if it is switched off when the GX device boots. This causes the On/Off/Chg/Inv indicator GuiMods adds to the Multi tile to be blank but you can still get to the details page to turn the inverter on.

Bucket1616, 2024-06-16:
Wow, super fast fix.Thanks for being such an asset to the Victron community!Mike

Bucket1616, 2024-06-21:
I think there may still be a lingering bug. The inverter disappeared after a few days. I created a video to show you what I am seeing. I also point out another bug with the tank level display at the bottom of the flow page. [https://drive.google.com/file/d/1A4Jm1JQ-T9Q9dZLzNXVgtkXOiNy3ildF/view?usp=drivesdk](url)
You’ll probably want to open a separate issue for that. 

Let me know if there is anything I can do to help you debug either.

Mike

kwindrem, 2024-06-21:
I see you are using the DC or AC coupled flow overview. Does the same Multi disappearance show in the GuiMods Simple display? That will help track down the issue.

Regarding the tanks: see if these blank tanks are showing up in the device list. If so, this indicates a tank that was detected but is no longer connected, or possibly reconnected as a second device. My code builds the tanks and temps based on what is on dbus which is also what shows up in the device list. Another check would be to go to the device list then back to the overview page. This will rebuild the list of tanks and temps.

Bucket1616, 2024-06-22:
I changed the display setting to 'simple' and the inverter block was still not present.  Once in Simple mode, I then turned off the inverter and the invertor block remained, showing as 'off', for about 5 seconds, and then the inverter block, and AC loads and shore-power blocks all dis-appeared.  In summary, the behavior seems to  be the same regardless of if in 'Simple' or 'DC Coupled' display mode.

As to the tanks, they are still being seen; they are from a Yacht Devices 4 in 1 module.  If you look at the linked video, you will see that one of the missing tanks display attempts to be shown when I drag the display to the left.  It seems to indicate that it is still there and may be a rendering issue?  That being said, I did have to replace that Yacht Devices device, which residual tanks remembered somewhere in the Victron OS.  When I replaced the device, the current tanks now show up at device #'s 9, 10, 11, 12, which causes they to display in the wrong order and they sort alphabetically, where 10, 11 and 12 come before 9 (because it is not 09).  Searching online has not allowed me to discover how to clear-out the remembered, past devices.  Not sue if there is anything to that piece of information, but wanted to make sure you have as much information to work with as possible.

Please let me know how I can next help,
Mike

drtinaz, 2024-06-22:
> I changed the display setting to 'simple' and the inverter block was still not present.  Once in Simple mode, I then turned off the inverter and the invertor block remained, showing as 'off', for [...]

To remove old devices, you should see 'remove disconnected devices' or something along those lines in the devices list (main menu screen). If you disconnect the tanks module and then delete all disconnected devices, when you reconnect the tank module it may then reassign the 4 tanks to the first available device numbers which should be the original numbers from the old tank module as long as you haven't added new devices since then. 

As for the tanks not being in the correct order, you can go into the individual tank setup and assign a custom name and this might reorder them.

Bucket1616, 2024-06-23:
Thanks for the pointers, however, I’ve tried this and it appears that even though the disconnected devices are deleted, there must be a persistent log that is remembering the device “VRM Instance”. I’ve tried the exact steps you suggest and the tanks come right back to the same device numbers. This kind of makes sense to me, as it is very conceivable that certain drives may not always be powered. If that always caused a renumbering for tank IDs, etc that could cause bedlam. Unfortunately, I’ve not found a way to delete that permanent log so that it can start over. As to renaming the tanks, this is also not an option since the four tank senders are on one NMEA device. The only option to give a custom name is under VE.Can devices. When I’ve tried to rename, only the first tank (tank ‘A’) is renamed, which leads me to believe that this is a limitation within the Venus OS. Mike

kwindrem, 2024-06-23:
The tanks display in the overview pages builds the tiles based on what services are reported by dbus.

It is possible your tank system is registering tanks on dbus that are not active. If after you remove disconnected devices you still see tank services in the device list then that's most likely what is going on and there really is not anything I can do about it.

kwindrem, 2024-06-23:
I released v10.47 which has some additional changes to show/hide the multi icon. I found a typo that was showing/hiding on the incorrect parameter. I also changed when I rediscover devices. Previously this was only done when switching to the overview page from the Menu pages. Swiping between overviews or going to and returning from a detail page did not do a refresh. Now the refresh is done any time the overview goes active (swiping between overviews, returning from a details page).

Bucket1616, 2024-06-23:
Thanks, the tank adapter is working right. Sometimes it displays correctly, other times it does not. All four tanks always show up if I am looking in the menu, it is only on the flow pages where it doesn’t look right. I have the “Tanks” page turned on and that always looks right, it is just at the bottom of the flow page, *sometimes* it gets screwy. Maybe not worth the trouble with GUI2.0 coming out soon. Mike

kwindrem, 2024-06-24:
I've looked over my tanks code and don't see why this might be happening with the other two pages displaying tanks properly. My flow overviews use the same code as the mobile overview.

If you can send me the log files when you are seeing this problem there might be a clue there. If you have ssh access, the log of interest is for the GUI:

/data/log/gui/current
or
/data/log/start-gui/current
depending on your firmware version

Please zip the file then post here.

Use the package manager backup settings to removable media and you'll find a logs.zip file. Post that here.

Also firmware version and GuiMods version.

Bucket1616, 2024-06-24:
Great!  I’ve installed it and so far, so good! Mike