kwindrem/GuiMods #256: Not all temperature sensors and solar charger shown

peterscarsten, 2024-06-16:
Hello,
after updating to 10.42 this morning, I can't see all my solar chargers (I have twonin place) and all my temperature sensors in the overview.

I will add some screenshots. 

Best regards 
Carsten[image]

[image]

TGTemplin, 2024-06-16:
I am having the same issue. My inverter temp will show, but none of my Ruuvitags. And I can only see the total solar. However, the hidden inverter is back.

kwindrem, 2024-06-16:
GuiMods v10.43 should fix these problems

peterscarsten, 2024-06-16:
Yes, it's fixed. Thank you

closed 2024-06-16 by peterscarsten