kwindrem/GuiMods #258: Request - Ability to touch temperature tiles/readings on Ekrano main pages screen to change relay thresholds

GTSR55, 2024-06-21:
Hello Kevin, first of all kudos to you for all your fine work and taking the Victron experience from great to excellent. Thank you!

Since I use my Ekrano GX relay function to act as my air conditioner thermostat in my RV that I live in full-time, I wonder if there was a way to touch on the temperature readout on the main screen and have it jump right into the menu screen where you change temperature values.

kwindrem, 2024-06-21:
I don't think this is a possibility. This section of the display already uses swipes to access additional devices and adding a touch region over the top of such a small dynamic area seems iffy. I do put touch regions over the top of STATIC tiles on the overview page to access the details pages but this is different in that the tanks and temps tiles are build dynamically.

GTSR55, 2024-06-21:
Here they are .....

GTSR55, 2024-06-21:
Hi Kevin,

Thanks for getting back to me so quickly.

I've attached pics of three of my screens for your reference. I see what
you mean regarding the main page, but perhaps the touch function could be
put on the second page where the tile is larger? Or perhaps even the third
screen?

I would gladly pay you for your time because I very much value programming.

The whole reason why I'm doing this is work around the Truma Aventa ECO air
conditioner "fan always on even when compressor is off" issue, which
results in sucking in all the humidity from the outside into the RV.

Myself and others apparently as I have read online, have contacted truma
more than once and I have no intention of making any any changes or or
improvements on this.

So I've worked out a great work-around for this by having the Ekrono drive
a power relay which turns the AC fan on and off after the compressor and it
works very nicely and no more rise in the humidity in the RV.

I didn't have to modify the air conditioner in any way shape or form so it
doesn't void their warranty. I simply have a Potter and brumfield 12 volt
relay control the 120 volts ac right out of the circuit breaker for the AC
in the breaker panel.

This is also a nice "twofer" because with this workaround I can now control
the air conditioning remotely wherever I'm at when I'm not in my RV.

Blessings

Gary

Show quoted text
Here they are .....

Show quoted text

kwindrem, 2024-06-21:
The tanks/temps/digital inputs page is also dynamically built so I'm not sure how to associate a touch area with a specific temp. I can look into possibilities. The other thought is that I don't have a way to associate a temperature with a temperature alarm which is what is triggering your "thermostat" functionality. Something else I'd need to look into. I do have a "relays" page and know which relay is an alarm relay so might be able to link from there.

Keep in mind that gui-v1 and threfore GuiMods is reaching end of life. Victron is getting ready for gui-v2 and has stated they plan to drop gui-v1 on all platforms except the CCGX.

GTSR55, 2024-06-21:
I am new to guimods as I am new to victron.

I have replaced my Xantrex inverter and everything except battery and solar
panels with Victron on my 2022 Winnebago Micro Minnie 2306BHS FLX.

I I must say Kevin that you have obviously set this move the bar and set
the standard very high for vectron as far as their interface is concerned.
When I found out about gooey mods and loaded it I was elated! Obviously
victron has incorporated most of your ideas into their upcoming V2.

Frankly, if I were them I would hire you as one of their chief software
engineer architects.

The fact that victron officially supports your work is not only a testimony
to your excellence but it says a lot about them as well.

When I read your announcement about not supporting V2, I sensed a bit of
frustration as I read between the lines. Please know this, your talent is
sorely needed in the community, and no matter what they build into v2, I
know that your talents, if you desire to offer them, will always be needed.

Thanks for the heads up on the relay function, I will research that and see
if that works for me. What I have now worked for me because I don't have to
change the temperatures too much for the AC to provide comfort in my rv.

Again one thing one of the things that was key to this project was the
workaround for the ridiculous controls on the truma air conditioner. They
make beautiful hardware this RV is all Truma except for the GE refrigerator
unit. But I called truma US and all they do is give you the brush off about
the air conditioner controls and whatever. They even told me the "inet
ready" label was absolutely bogus and useless on the CP Plus control unit
that controls the heater and air conditioner and there's no intention of
changing it they said.