kwindrem/GuiMods #260: Cannot install v10.51 - "GuiMods install failed could not patch some files"

thespinmaster, 2024-07-07:
Hi, 
I had the latest beta version of the Venus OS installed version v3.40~34 (Large Image) on a Cerbo GX (MKII)
Along with v10.51 of GuiMods installed and running ok.
I then proceeded to revert back to the "Official Release" v3.34 (Large Image)
After which I can no longer install and run GuiMods. 
The UI shows the following error "GuiMods install failed could not patch some files"
Have tried removing and re-installed GuiMods without success.

kwindrem, 2024-07-07:
GuiMods v10.51 installs properly on my RPI test system running v3.34. (SetupHelper is v8.6.)

You might trying downloading GuiMods. And update SetupHelper if it's out of date.

Otherwise I'll need logs to diagnose further. See the SetupHelper ReadMe for how to retrieve the logs.

thespinmaster, 2024-07-07:
I could not get rid of it, so I downgraded the Venus OS to 3.14 and then installed an older release of GuiMods. Then upgraded back to the latest versions of both. All good now. Thanks

closed 2024-07-07 by thespinmaster