kwindrem/GuiMods #261: Guimods

Gexle, 2024-07-09:
Hello

I updated to the latest Guimods version and now the display is white and I can't see anything anymore.

What should I do now??

How can I get it back??

[image]

kwindrem, 2024-07-09:
Reboot if you have not done so already.

Do you have ssh access? If so:

/data/GuiMods/setup uninstall auto

logs would also be very helpful.

tail -50 /data/log/PackageManager/current  | tai64nlocal
tail -50 /data/log/gui/current  | tai64nlocal
tail -50 /data/log/start_gui/current  | tai64nlocal

What firmware version?

Gexle, 2024-07-09:
The Firmeware is V2.93 is run
restart didn't help!

Gexle, 2024-07-09:
Where can I find the logs again?

Gexle, 2024-07-09:
I can't find start_gui!

[logs.zip](https://github.com/user-attachments/files/16146136/logs.zip)

kwindrem, 2024-07-09:
v10.54 is available now that should fix the white screen. Please set branch/tag back to latest if you changed it.

Thanks for reporting.

closed 2024-07-09 by kwindrem

Gexle, 2024-07-09:
Thank you, everything is working as before

Thank you