kwindrem/GuiMods #262: scrambled GX page

Snukkel, 2024-07-16:
after updating to V3.40-37 and/or 39 i get scrambled page in my browser.
after downgrading to V3.40-34 it works fine .
also after removing guimods, page is normal again.

Upgrading to V3.40-39 and reinstalling Guimods, page is OK, but after some time page is scrambled again.

System running on GX.

[Remote Console on LAN.pdf](https://github.com/user-attachments/files/16245798/Remote.Console.on.LAN.pdf)
[image]

kwindrem, 2024-07-16:
You say "GX" but not which one. Cerbo?, Ekrano, CCGX??? Which firmware version?

Is the browser running locally or via VRM or ???

Since it works  for a while my guess is the style sheet is getting corrupted but nothing my packages do would change it after install. Are you installing a mod to the style sheet such as hiding the hot keys?

I'd need logs from the system to help further. SetupHelper's ReadMe explains how to collect logs.

Snukkel, 2024-07-16:
> <https://github.com/kwindrem/GuiMods/issues/262#issuecomment-2230986249>view
> it on GitHub, or

It is Cerbo.
firmware was in attached PDF file and in the text.

normally  Browser, VRMis also weird.

GUIMODS simple layout is OK , but in a log page 
with the big black Icons as in the PDF.

Rgds
Peter Sibbing

Peter Sibbing
Beukenlaan 3
6741 KB Lunteren
The Netherlands

Tel:  +31-318-487077

E-mail: ***@***.***

kwindrem, 2024-07-16:
Reproduced here. Working on it.

kwindrem, 2024-07-16:
Fixed. You need to update SetupHelper to v8.8 to restore the remote GUI.

closed 2024-07-16 by kwindrem

Snukkel, 2024-07-16:
UPgraded to 8.8, no change.

rebooted,  no change.

Cold reboot, No change. still scrambled.

kwindrem, 2024-07-16:
Uninstall GuiMods and reinstall it. That should force the style sheet file to be replaced. The issue is I compare the installed file with the one I will replace it with and don't do anything if it's the same.

Snukkel, 2024-07-16:
Ok, thanks.
did as stated below. it is working again.

Rgds
peter

kwindrem, 2024-07-16:
Great. I'll fix SetupHelper so that the uninstall / reinstall isn't needed.