kwindrem/GuiMods #263: Rights styles.css

ikorpershoek, 2024-07-16:
I noticed after installing the latest version the remote console on the local LAN output was totally garbage.  Finally I found that the /var/www/venus/styling rights are changed to 600  I changed it back to 644 and everything is OK again.  I could reproduce this several times. Maybe it's important

kwindrem, 2024-07-16:
Fixed in v8.8 of SetupHelper

closed 2024-07-16 by kwindrem

xsnrg, 2024-07-16:
8.8 did not seem to fix the console for me. I even forced a reboot after installing

kwindrem, 2024-07-16:
Turns out SetupHelper would not reinstall the style sheet file or change its permissions if it was already installed. I'm releasing SetupHelper v8.9 to fix that. Our you an uninstall GuiMods and reinstall it.