kwindrem/GuiMods #264: Battery Detail missing details.

angyp, 2024-07-22:
Hi, I seem to be missing some details on my battery detail screen. Ie Consumed and  Remaining 

[image]

When I look at the BMS details I see the missing info.

[image]

Thanks

kwindrem, 2024-07-22:
I only have a dumb battery so I a not debug this issue. I need to see the battery service in your system reported in dbus-spy which is a command-line utility

kwindrem, 2024-07-29:
I just released some changes to the battery detail (GuiMods v10.60). Please check to see if you now see consumed and remaining values. Thanks

chelle-hine, 2024-07-29:
I think this update has broken the display of Consumed (Ah) on the 'Battery detail' tab, when using a Smart Shunt. Today my Cerbo stopped showing Consumed (Ah) and instead shows an empty 'Aux Voltage' slot, even though I am using the Aux input on the shunt for Temperature. The 'Temperature' slot is displaying values correctly. I checked the package manager and my Cerbo has auto updated to V10.60, and was displaying fine before the update.

angyp, 2024-07-29:
Works for me.

[image]

kwindrem, 2024-07-30:
I thing you should also be seeing remaining AH. Made another change (v10.61)

chelle-hine, 2024-07-30:
@kwindrem v10.61 fixed my issue, thanks
[image]

kwindrem, 2024-07-30:
There is still something wrong. The battery details page is showing both installed and remaining capacity so my details page should be including consumed and remaining. Remaining AH should appear next to Consumed AH.

kwindrem, 2024-07-30:
chill-hine do you have a smart battery or a battery monitor/shunt? If shunt then that explains why there's no remaining value shown as installed capacity is not provided. let me know. thanks.

chelle-hine, 2024-07-30:
> chill-hine do you have a smart battery or a battery monitor/shunt? If shunt then that explains why there's no remaining value shown as installed capacity is not provided. let me know. thanks.

I have a Smart Shunt and dumb batteries. I've also checked that when discharging 'Remaining time' shows OK. As you say, Shunts don't output the Remaining Ah, just time.

kwindrem, 2024-07-30:
I'll close this for now then.

closed 2024-07-30 by kwindrem

angyp, 2024-07-30:
Something is still not right.

[image]

[image]

kwindrem, 2024-07-30:
Made another fix v10.62

angyp, 2024-07-31:
That works. Thanks

[image]
[image]