kwindrem/GuiMods #272: Link to external running state information needed (docs)

Gregrvr, 2024-09-16:
Hi Kevin, I'm using GUIMods 10.64 with Venus OS 3.41 on a pi in my RV. I also have Home Assistant running on the RV and built a simple Remote Relay control to start my built-in generator. I have written up an extensive post on [Victron's Community Site](https://community.victronenergy.com/t/cannot-publish-generator-state-with-mqtt-on-venus-os/1798/10?u=sungunners) and have learned much in the last few days. I've realized using the excel spreadsheet there are only a few writable state conditions. You have a switch called Link to external running state. Reading through your code, it's not clear to me what exactly this does or what input it uses to sense when the external generator is running or if this input is digitally writeable. This is explained in detail in the post referenced above. BTW, there are a few typos in the comments in /GuiMods/FileSets/PatchSource/dbus_generator.py.patch. I would be more than happy to fix them.

I cannot run a physical wire between the generator controller and the pi. I am relying on MQTT to digitally write the states to the Cerbo. I'm hoping this feature in GUIMods will allow me to do this. That will tie both systems together. Is there I topic I can digitally write 1 when the generator is running and 0 when it is not?

In manual mode, everything is working great! I'm using the ManualStart = 1/0. But RunningByCondition is the problem. I'm trying to let the Cerbo know the external generator is running. If there isn't a way to do this, then I can only use RunningByCondition with the Link to external running state off to start/stop the generator without feedback to the Cerbo when AutoStart is enabled.

kwindrem, 2024-09-16:
GuiMods changes a few things in the generator interface:

I synchronize the internal manual run with the state of the generator digital input. That is if the digital input changes state then the manual run state is brought into agreement with the digital input. This allows someone to start the generator externally and have manual run activate or to stop it externally and have manual run go inactive. 

If an auto run condition is active commanding the generator to run, I do not manual run to the digital input. In stead, I show a warning on the generator overview indicating that someone stopped the generator externally when is should be running based on the automatic run conditions.

If a digital input is not programmed as generator, I use the the AC input frequency if the AC input type is generator to trigger sync and "external override".

In addition, I prevent run time accumulation of I detect that the generator is not running.

All of these are qualified by the Link to external running state preference.

There is no way to set a digital input state externally. It's all handled by systemcalc's relaystate.py. The dbus parameter is read only.

The .patch file is automatically generated so should not be edited. If you can identify the typos I'll see if I can update the original code from which it is generated.

Gregrvr, 2024-09-16:
Thank you so much for your reply! Your reputation is outstanding!

I just tried publishing (successfully) to /system/0/Relay/0/State using values 0/1. If I understand you correctly, this should be the digital input you use to sense the externally running generator that you referred to? Too late for me to test tonight.
[image]

I'm happy that I chose Victron to build my solar system on my 2016 RV.  I just think it's important that they system gets feedback (when possible) that the generator starts/ stops/ and is actually running when it's supposed to be. My built-in generator did not have that capability until now. This is what I'm trying to accomplish digitally with my simple controller.

**Typos**
Line 12 A switch in the generator settings **_menu controls_ _whether_** the incoming state affects manual start or time **_accumulation_**
Line 14 for **_automatic_** start / stop

kwindrem, 2024-09-17:
The relay state is for the OUTPUT of the generator. That is, it causes the generator to run or stop.

The parameter you want is for the digital input defined as generator. This will create a dbus service starting with com.victronenergy.digitalinput. With a type of generator. That services /InputState and /State parameters are read only but you can change this by adding writeable=True to the line that creates the parameters in /opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py.

After making them writable, writing to /InputState doesn't seem to do anything but writing to /State does seem to work: With GuiMods installed, changing /State between 11 (stopped) and 10 (running) causes the manual running state to change in the same way as changing the physical digital input state.

Note that if the digital input state actually changes, then your setting will be overwritten. Also a reboot will require you to set it again.

Gregrvr, 2024-09-19:
Before delving into changing any code, I did a little experimenting today. I'm not finished yet and won't have any more time until this weekend. I did install RpiGpioSetup which gave me a few more relays to work with. I tested all the relays by taking volt readings from all the relays by toggling them from the Relay page on the remote console. They do go high when on and low when off. I then selected relay 6 (GPIO 18) and jumpered it to Digital Input 1 (GPIO 29). I then published 0/1 to victron-rv/W/device_id/system/0/Relay/5. Writing 1 Turns on relay 6 on the relay page and 0 turns it off. I then added this to my start cycle and when the generator state changed to On, I got the timers to work! I did have to invert the writes. 0 Turns on the counters and 1 Turns them off. Not sure why this is. I did not have time to try any AutoStart conditions. I ran out of daylight to think about that. Just an update...needs more work.

kwindrem, 2024-09-19:
Yes you can certainly loop back a relay to a digital input. I do this for testing. The reason you needed to invert things is because the digital inputs are active low and the relay outputs are active high. For my testing, I simply invert the digital input.