kwindrem/GuiMods #274: Problem with the presentation

OEBliss, 2024-09-25:
[image]
Everything works very well except for one thing:

When I switch off the inverter, the displays on the right and left disappear after a few seconds and there is a question mark on the inverter.

The displays should actually stay on and the inverter should say “off”.

What could be the reason for this?

kwindrem, 2024-09-25:
The AC input tile should disappear but the question mark in the inverter tile is not correct. Check the device list to see if the inverter is shown. If not,the GX device can not determine the inverter state or anything else about the inverter.

In the stock code, Off is displayed in the top line of the inverter tile even if no inverter is found. And the AC input tile is always shown as well.

Rebooting the GX device might help but not a certain thing.

kwindrem, 2024-09-25:
Correction. It apparently does now behave as you describe. This is new and not inteneded and I’ll look into it next week after I’m back from vacation.

OEBliss, 2024-09-25:
Perfect, thank you very much for your efforts!

kwindrem, 2024-09-29:
The problem is intermittent. I was just on a trip where my RV was off grid. Sometimes the inverter tile would show Off and other times it would show ?

I can not reproduce this yet but will keep trying and when the ? appears again I can hopefully figure out why.

kwindrem, 2024-09-30:
I was able to reproduces this problem by switching between ON and INVERT ONLY then to OFF. It also happened if there was no grid connection when switching between ON and OFF.

I was able to fix the problem by updating my Multi from v508 to v552 (current at the time of this comment).

Please let me know if you find this not to be the case. 

The problem appears to be related to some of the Multi's dbus parameters going undefined. This causes the GX device to no longer have a valid com/victronenergy.system /VebusService parameter.

kwindrem, 2024-09-30:
I spoke too soon. It's still happening.

kwindrem, 2024-10-01:
I am afraid I am not going to be able to fix this problem The cause is that /Connected goes false (0) even though the device is still connected and communicating. This causes systemcalc to remove the Multi from its calculations. Attempts to find an alternate way to detect the Multi have been unsuccessful. I'm afraid Victron is going to have to fix this one.