kwindrem/GuiMods #275: Can't install v10.66 Rechecks are failing

Jens-dev, 2024-09-25:
I try to install via packageManager but getting the following error:

ERROR: patch error details were saved in GuiMods/patchErrors
install failed during prechecks - no changes were made
style.css could not be patched

It is a clean installation on a more or less fresh system. Only some other packages are installed.
I also tried to uninstall and install via ssh, but with the same result.

kwindrem, 2024-09-28:
I have been able to reproduce this and working on a SetupHelper fix. This error only occurs when the package is already installed.

In the mean time, a work around:

uninstall the package then reinstall it.

If doing this from the command line include uninstall on the command line so that checks are skipped:

/data/GuiMods/setup uninstall

kwindrem, 2024-09-29:
I have resolved this issue. A new version of SetupHelper (v8.21) is required for the fix.

closed 2024-09-29 by kwindrem