kwindrem/GuiMods #277: Relay colour button change on operation for conformatley

RobDuthie, 2024-09-28:
File name: TileRelay.qml

Changed these colours to make as per standard switch orietation for safety and much easier to see what postion it is in etc.

baseColor: !darkMode ? (onButtonActive ? "green" : "#e6ffe6") : (onButtonActive ? "#fe2400" : "#7d7d7d")

baseColor: !darkMode ? (offButtonActive ? "black" : "#e6e6e6") : (offButtonActive ? "green" : "gray")

kwindrem, 2024-09-28:
Thanks. I am making some changes to colors to make things more obvious based on your input but didn't want to make on = danger (red) nor off = safe (green) in all cases.

RobDuthie, 2024-09-29:
Need to write a scripe file to stop the undate changes, as the red and green is much better in dark mode,  Try it on thre next rewrite, as its only for the relay controls.?