kwindrem/GuiMods #278: incomplete file set for v3.50-24

dsrawding, 2024-09-28:
Can't install v10.66 due to incomplete file set for v3.50-24 error

kwindrem, 2024-09-28:
Major changes from Victron. Working on an update but it will take some time.

kwindrem, 2024-09-28:
should be fixed in v10.67

closed 2024-09-28 by kwindrem

dsrawding, 2024-09-28:
Updated to v10.67 and still getting the same error unfortunately.

dsrawding, 2024-09-28:
I removed from the gui, logged into the cli and removed the directory.  Then downloaded the code from GitHub and still got the same error.  Looked in the FileSet directory and both ~24 and ~25 were there.  System was still trying to install from ~24.  Removed ~24 and got it going.  

Stange though as I then downloaded the code again a couple minutes later from GitHub and ~24 was gone.  You were probably working your magic.  Anyway I'm up and running!

kwindrem, 2024-09-28:
There should not have been a ~24 directory. I never created a file set for v3.50~24. Could have been left over from a previous install attempt but the entire /data/GuiMods directory should have been replaced.

Any way, glad it worked.