kwindrem/GuiMods #280: How can I add a new page to the GUI to control some local stuff on lan with HTTP requests?

pbawesome, 2024-10-16:
Either that or with integrations into node red, total beginner here, just looking through all the qml files and have no idea what I'm looking at so far.

jamiejetta1986, 2024-10-16:
I'm delving into node red at the moment, I'm trying to get some Shelly plus 1's to talk via Mqtt to switch them on and off via relay controls on the rpi

pbawesome, 2024-10-16:
Controlling the cerbo relays in node  was very straightforward. Let me know if you need help. Any ideas on the custom page?

kwindrem, 2024-10-16:
I added a couple of overview pages from scratch and that wasn't terribly difficult. Look at my OverviewRelays.qml page in GuiMods. But Node Red might be easier for someone that has not yet gotten familiar with QML and the Victron code base. It will also work with the new GUI. No way yet to make QML code modifications that will work on both the local and remote GUIs.

jamiejetta1986, 2024-10-17:
> Controlling the cerbo relays in node  was very straightforward. Let me know if you need help. Any ideas on the custom page?

I mean am I doing the correct thing here? 
do I need to use the relays or can I use digital outputs to control the Mqtt? 
This is new to me so I'm not sure on the best way to do it...

I haven't had any time this week to do any research. But will be looking into it this weekend, 

i have a complete rpi setup in my house as a test rig, if I can get it to work on that, then I can replicate what I learnt onto the van unit...