kwindrem/GuiMods #282: GuiMods v10.68  not working on new CerboGX firmware v3.50

kocki2, 2024-10-31:
GuiMods v10.68  not working on new CerboGX firmware v3.50

Tadpole48, 2024-10-31:
Agreed, "incomplete fileset for v3.50"

RobDuthie, 2024-10-31:
As Kevin said stay on ver 3.42 as there may not be any updates for guimods due to the new UI  from Victron.
As it is not compatible with raspi units software, As Victron are going to stop updating supporting there classic UI.

kwindrem, 2024-10-31:
Sorry for the delay in getting GuiMods updates to work with v3.50. My personal time is focused on getting into a new home at the moment but I will be getting back to GuiMods shortly.

I will continue to support GuiMods for gui-v1 as long as it is offered as an option in the firmware. After that, I must abandon the bulk of GuiMods as it's totally incompatible with gui-v2. I am looking at options to support portions of GuiMods to enhance the generator interface.

Tadpole48, 2024-10-31:
Many thanks, I appreciate the difficulties with GUI V2, I just want access to my external relays (even through Node Red will do), can anyone point me in the direction of possible access to external relays in GUI V2? and if that's even possible?

RobDuthie, 2024-10-31:
Need to  update the PageSettingsRelay.qml file thats gets you back the relay control, but no front page button on front screen as gui mods need to be updated with new packages, or being installed separately. Kevin will  update later as asdvised above
The analog ports are all OK still.

Tadpole48, 2024-10-31:
> Need to update the PageSettingsRelay.qml file thats gets you back the relay control (Just tested OK now), but no front page button on front screen as gui mods need to be updated with new packages, or being installed separately. Kevin will update later as asdvised above The analog ports are all OK still.

Many thanks, I'll take another look at the qml file.

kwindrem, 2024-11-02:
I have a version of GuiMods working for v3.50 and v3.51~2. 

Let me know if you experience any issues.

Again note: GuiMods menu enhancements will not appear in the New UI (aka gui-v2).

You will however be able to configure the external transfer switch and link generator external run to manual run by switching back to Classic UI. 

This also applies to Package Manager. If you configure Package Manager for auto downloads and installs then these will occur in the background even though you won't have those menus.

kocki2, 2024-11-03:
[image]
It's working, great and thanks.
Regards, Gerd

julian4646, 2024-11-03:
Thanks for working on this - will wait for 3.51-2 stable to be released - also getting below on 3.50:

[image]

kwindrem, 2024-11-03:
fixed in v10.73 for v3.50

julian4646, 2024-11-03:
> fixed in v10.73 for v3.50

thanks alot - working fine in VenusOS 3.50 (I rebooted to refresh packagaes available then it appeared and installed no issue)

CzarofAK, 2024-11-08:
> I have a version of GuiMods working for v3.50 and v3.51~2.
>
> Let me know if you experience any issues.
>
> Again note: GuiMods menu enhancements will not appear in the New UI (aka gui-v2).
>
> You will [...]

I run v3.50. Without Gui-Mods, and Lucifer06 "RemoteGPIO" i have only one Relay shown. If I run both mods, i get two relays shown in the GUIv2 (not four, but one additional)! so, it does something. Is it, because Victron has max. two relays in their systems?

kwindrem, 2024-11-08:
GuiMods shows relays native to the GX device plus those configured in RemoteGPIO in the relay overview and also in the relay configuration menus.

If you have not added an external relay module and configured i in RemoteGPIO then you will only see the relays actually present on the specific hardware running Venus OS. For most Victron devices like Cerbo, that's 2 relays.

Please look at the documentation for RemoteGPIO which is supported by another author for further information and post an issue there for additional support.

CzarofAK, 2024-11-08:
It was not a "bug" report rather than something that i found curious.
I have only a Rpi as testsetup, usually showing one relay. I thought i give it a try to see if i can enable Guimods and RemoteGPIO with the new GUIv2.
As long as i load the old Gui its all fine. In the new gui i expected to se no change at all, but there is one additional appeared.

kwindrem, 2024-11-08:
Not sure how gui-v2 determines which relays are in the system.

On the PI, the stock gpio_list is empty or does not exist. RpiGpioSetup (my package) populates up to 6 relays and 5 digital inputs, however RemoteGPIO and RpiGpioSetup are mutually exclusive as they both modify gpio_list, so when RpiGpioSetup is uninstalled as part of the RemoteGPIO install, the 6 relays are removed.

Check with the author of the RemoteGPIO package to see if he has an explanation.

closed 2024-11-08 by kwindrem