kwindrem/GuiMods #288: GuiMods documentation confusing under firmware 3.52

davinsoftware, 2024-12-18:
Can someone explain what is going on with GuiMods once you upgrade Victron firmware to 3.52? The README is quite confusing. 
It says: 
"starting with firmware v3.50, the "New UI" (aka gui-v2) is available for use"

Is "New UI" the new GuiMods UI or the new Victron 3.52 native UI? 

Also at the end it says: 
"Note that installing GuiMods will uninstall GuiMods2.
Installing GuiMods2 after installing GuiMods will uninstall GuiMods.
You should not use GuiMods2 !!!"

So we shouldn't use GuiMods2, but "gui-v2 is available for use." I believe GuiMods2 is some old deprecated GuiMods package, which just happens to be named almost the same as "gui-v2." Correct?  And gui-v2 is V2 GuiMods, or V2 Victron? 

I just updated firmware to latest and GuiMods to latest (I'm behind because I do manual updates). 
When I do the UI switch to Classic UI, it looks like I get the old Victron UI but cannot access GuiMods UI. Am I missing something? GuiMods had some trouble auto-updating via Package Manager (install error number 1) but eventually succeeded and rebooted.

How do I get back to GuiMods if I still want to use it?  Or is GuiMods being disabled / deprecated in 3.52+ firmware?

kwindrem, 2024-12-18:
The "new UI" is gui-v2, the new STOCK Victron user interface. It has nothing to do with GuiMods which is an enhancement to the original stock user interface.

GuiMods will not enhance gui-v2, the new stock Victron user interface but will enhance the original stock (aka Classic) user interface.

GuiMods2 is in fact an old, short-lived version of GuiMods which is obsolete hence the excerpt from the ReadMe that you quoted.

There has been a recent change as part of firmware v3.60~2 and beyond that prevents GuiMods from installing which is why you see the original un-enhanced classic user interface with that firmware. I'm working on a fix but it's taking some time due to the extent of the changes, specifically renaming a directory for the generator interface which breaks SetupHelper and the utility I use to create firmware-dependent "file sets" within each package.

GuiMods should work with firmware v3.10 through v3.52. If it is not, insure you have the latest SetupHelper installed as there have been recent changes that rely on the latest version to install packages properly. If you are running the latest version of GuiMods and SetupHelper, then post details here. Logs are very useful in such situations and the SetupHelper ReadMe explains how to collect them.

davinsoftware, 2024-12-18:
Thank you. So I think yes I'm stuck in the state where "Classic" UI doesn't
bring me back to GuiMods. I have GuiMods 10.78 installed on firmware 3.52.
I tried to update SetupHelper from 5.10 to 8.22 via Package Manager but
install says "Unknown error 245". I will look into that more.

davinsoftware, 2024-12-18:
Oh, now Classic UI has got me back into GuiMods. Not sure what happened. Either the extra reboot did it or the extra SetupHelper failed update.

kwindrem, 2024-12-18:
You may need to uninstall SetupHelper then do a blind install for it to get you from 5.10 to the current version. If you have command line access, that might be easier.

closed 2024-12-19 by davinsoftware

davinsoftware, 2024-12-19:
Thank you. FWIW I think GuiMods is still better than Victron's v2 UI. So I am happy we still have it, and appreciate all the work you do.