kwindrem/GuiMods #290: Guimods on certo gx

kaskelot1, 2024-12-28:
I'm a newbie. On my cerbo GX MK2 I have Venus 3.52.
If I' m correct, I need to Revert back to 3.50 or earlier, to have full guimods .
Is it correct? And which release of guimods shall I have?
Best
Marco

SaVok0887, 2024-12-28:
Hi. Guimods work on 3.52 as well.

kwindrem, 2024-12-28:
My apologies for not getting GuiMods updated. Victron through me a curve in v3.53... by renaming the generator subsystem and I have had to rework some things to accommodate the change. 

As SaVok0887 says, GuiMods v10.78 should work on firmware v3.52. It's just v3.53 beta and beyond that are not supported yet.

kaskelot1, 2024-12-28:
Ok thks
This is my dashboard
[Immagine che contiene testo, schermata, software, Icona del computer  Descrizione generata automaticamente]

And my settings
***@***.***

Now I have only temp sensors. Other equipment are on board a boat.
How How can I have a different dashboard like this one?

[Supercharged Victron displays & push alerts - Electrical - SeaBits  Discussions]

Thanks
M

kwindrem, 2024-12-28:
Not sure what you are asking. I'm not seeing any images or other attachments. I did find a web page on seabits.com that shows the GuiMods display and this should be shown on your system with GuiMods installed and configured correctly.

You mention "dashboard" which suggests you are looking at a VRM page. GuiMods won't show there as it is a completely different user interface.

kaskelot1, 2024-12-28:
ok. i'm on remote console
this is a screen shot
[image]

and settings
[image]

 How can I have a different dashboard like this one?

[image]

thks

kwindrem, 2024-12-28:
Your first screen shot appears to be the enhanced mobile overview page provided by GuiMods which indicates it is installed and functioning. If you swipe left or right on the page it will scroll through other available options. One should be the main overview page like the one shown in your third screen shot. (This scrolling behavior is part of the stock Classic user interface, not added by GuiMods.)

GuiMods provides options to show or hide various of these pages so you can narrow the options to the ones you prefer to display.